motioneye-project/motioneyeos #2077: foscam and use MotionEye OS as a dvr?

bob5731, 2019-09-02:
can I add foscam and use MotionEye OS as a dvr?

bob5731, 2019-09-02:
I have foscam fi9800p. I now have my camera imported but can't see any video what did I do wrong? Is just sitting there spinning.
I believe I'm using the same foscam camera as mentioned above. What would the URL look like that I need to enter just to make sure I got it right?

mb39924, 2019-09-03:
are you using chrome? of so, update to the latest nightly build (EExpert settings > Enable Prereleases > check for updates)

bob5731, 2019-09-07:
Using raspbian

bob5731, 2019-09-12:
It is an issue with chrome. How do I fix it in chrome?
My next question is if I hook an external 1tb hard drive up to it how do I set it up?