motioneye-project/motioneyeos #2078: Pi Zero not starting Motioneye from motioneyeos

Zach0331, 2019-09-02:
Hello,

I have grabbed motioneyeos-raspberrypi-20190427.img from the repo. I flashed it to the SD card using Etcher. I created my wpa_supplicant.conf file and ejected my SD card from my laptop. Then, I inserted it into my pi zero and the camera is installed on it too. I connected a monitor to it and plugged in the power cord. The system finishes everything correctly until it gets to "starting motioneye". That part always fails and reboots the system. I have the camera installed correctly from looking at guides and also comparing to my PI3. I have no clue what else to do. Any help would be greatly appreciated.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20190427.img

### Board Model

I am using the following board/model: Raspberrypi Zero W

### Camera

I am using the following type of camera: Pi Camera for Zero

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI

johnfc2018, 2019-09-13:
Start by fault finding:
1. Delete the wpa_supplicant.conf file from the memory card. If the system boots okay, you either have a misconfigured wpa_supplicant.conf file or it has extra symbols at the end of each line. 
2. Burn the latest version of motioneyeos-raspberrypi without removing the xz from the end. 
3. Make sure the memory card is big enough for the entire image.
4. Make sure the power supply is delivering enough power for both the pi and camera, as it may not be using full power until the software loads.
5. Make sure the camera connector is inserted properly.

Hopefully you will find the answer.

ThomasManthey, 2019-09-29:
Maybe you problem is related to my because I want to run the camera autonomously with an auto connecting to my mobile hotpot if it is in a reachable range for getting the data. https://github.com/ccrisan/motioneyeos/issues/2108
You could check if the gui interface setting for wifi is made correct. The setting in wpa_supplicant is no used as I found out.
The best would be to connect you pi to a monitor an a keyboard to configure wifi.