motioneye-project/motioneyeos #2080: Motioneyeos raspberry pi zero W configuring wired network: no device

maxpi13, 2019-09-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry pi zero W and raspberry pi 3 b+

### Camera

I am using the following type of camera: Fast Network Camera

My camera model is: Raspberry Pi Camera V2.1

### Network Connection 

My motionEyeOS unit is connected to the network via: Ethernet for hub server and wifi for camera

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Monitor (no wifi adapter as raspberry pi Zero W has one built in)

### Log Files

I consider the following log files relevant to this issue:

 
[image]

maxpi13, 2019-09-02:
Ok so this is my first time posting here and it is because my setup with a pi 3 (as hub) and a pi zero W (as cam) is not working. I have made sure the wpa_supplicant.conf file is setup correctly about 15 times now (and remade it a lot of times) and i have re flashed many times too. the error (see above) will not go away and the pi just falls into a reboot loop every time. My raspberry pi 3 (hub server) on the other hand appears to be functioning perfectly (albeit without any cams yet). Could i just buy a micro USB to Ethernet adaptor for initial setup and then in the online user interface change it over to WiFi? I will try to reply as fast as possible but I'm new to the whole programming thing so go easy on me with the technical terms! ;)

closed 2019-09-03 by maxpi13

angadsingh, 2020-12-02:
hey @maxpi13 did you find a solution to this?

sl4ppy, 2020-12-28:
You didnt remove the .txt extinsion on the filename wpa_supplicant.conf.  Windows doesnt show extensions by default, so the file is probably named wpa_supplicant.conf.txt

mdyeargin, 2021-01-16:
I have the same issue and have tired the blank SSH file as well as making sure that it does NOT have .txt by showing hidden file extension. 
[image]

mdyeargin, 2021-01-16:
Oh, forgot to provide hardware spec: Raspberry Pi Zero W ver 1.1 2017

starbasessd, 2021-01-16:
misspelled wpa_supplicant.conf

starbasessd, 2021-01-16:
ssh should be lowercase: ssh or ssh.txt

mdyeargin, 2021-01-16:
Yep, caught the spelling error, and also had tried the upper and lower (screen shot was of upper) as well as with and without .txt on the ssh. Finally give up and started all over, but this time didn't use text edit but notepad ++ to create and write the file wpa_supplicant.conf file, note I had rewritten the file five times and caught spelling on 2nd write using text edit (i.e. MS Notepad). Thanks for such fast response, got it working well.