motioneye-project/motioneyeos #2081: Brightness control is missing

OldHolborn01, 2019-09-02:
motionEye Version 0.41rc1
Motion Version 4.2.2
OS version motionEyeOS dev20190820
Camera Typ V4L2 Camera, no codec selection is possible

Since this version and one before I lack the brightness control. The pictures / video are heavily overexposed and I can not adjust anything. Automatic brings no improvement...

Please help

[image]

ccrisan, 2019-09-03:
What does the following command return?

    v4l2-ctl --device /dev/video0 --list-ctrls

OldHolborn01, 2019-09-03:
[root@motionEyeP3 ~]# v4l2-ctl --device /dev/video10 --list-ctrls

User Controls

  min_number_of_capture_buffers 0x00980927 (int)    : min=1 max=1 step=1 default                                  =1 value=1 flags=read-only

I just have updated the Image to motionEyeOS dev20190829, but the same.

I have a second Pi 3 with even the same image work and V4L2 Camera, this is the original Camera Modul V2.1, - this works fine with all options! The device is /dev/video0. But the first one ist a Camera with FishEye lense RPI-CAM-G ...dev/video10,

Both cameras/Pi´s show the same output of User Controls...

The Problem occours since update from earlier 20190427