motioneye-project/motioneyeos #2087: wpa_supplicant/static_ip on initial boot

rdb9879, 2019-09-04:
Greetings.

I normally do a "headless" install for motioneyeos onto a raspberry pi zero W. I utilize the "wpa_supplicant.conf" and "static_ip.conf" files on the SD card root to connect over wifi. I downloaded the latest version (motioneyeos-raspberrypi-20190427.img) and it just would not connect to the wifi network. I reimaged/retried at least 4 or 5 times. Not able to ping, and could not see the device in my gateway router's "connected devices" page.

I finally roll back to the previous version I had downloaded (motioneyeos-raspberrypi-20180224.img), did the same steps, and it worked right away first try.

Has something changed in these *.conf files that prevent the new version from utilizing them?

ccrisan, 2019-09-04:
Let's see some logs. Let's start with `boot.log`.

slurmpit, 2019-09-18:
Hi
I have same problem. 
How can I get boot.log if I have no network connection and access to file system?

gb-dursley, 2019-10-07:
I too have had this problem.  Traced it to a failure of boot to overwrite copies of wpa_supplicant.conf in /etc and /run.