motioneye-project/motioneyeos #2090: Raspberry PI 3A, dev20190904, Firmware rejected country setting, no WiFi

klaus-ditze, 2019-09-05:
Hi,
did try the  **_dev20190904_** for my Raspberry PI 3A.

Get no WiFi connection.

In the dmes.log i found:

```
[   26.621184] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   26.621309] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   26.621881] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[   27.091845] brcmfmac: brcmf_cfg80211_reg_notifier: Firmware rejected country setting

```

In the wpa_supplicant.conf the country=DE 

No Problem with the WiFi in the last release _**20190427**_

klaus-ditze, 2019-09-06:
Same Error with Prerelease **_20190904_** on a Raspberry PI 3+ 

```
[   26.839366] brcmfmac: F1 signature read @0x18000000=0x15264345
[   26.843516] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   26.844422] usbcore: registered new interface driver brcmfmac
[   27.003030] random: crng init done
[   27.003041] random: 2 urandom warning(s) missed due to ratelimiting
[   27.058543] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   27.058622] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   27.059032] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[   27.248417] brcmfmac: brcmf_cfg80211_reg_notifier: Firmware rejected country setting

```

klaus-ditze, 2019-09-06:
@ccrisan 

Found the Solution!

Before building the Raspberry PI 3 Image you have also to  put the right `brcmfmac43455-sdio.clm_blob` file  from `https://github.com/RPi-Distro/firmware-nonfree/raw/master/` to  `motioneyeos/tree/dev/board/raspberrypi3/overlay/lib/firmware/brcm` 

Please fix.

ccrisan, 2019-09-09:
Fixed in https://github.com/ccrisan/thingos/commit/96a19ddb3143e34b1c46001371fea002331c9356.

closed 2019-09-09 by ccrisan