motioneye-project/motioneyeos #2091: motioneye change wifi to another wifi

johannscholtz, 2019-09-05:
hi

i have a motioneye system working on wifi at my house, i want to change the wifi settings to use in another house.

how do i do it? how do i change the wifi settings. 

 i dont see wpa_supplicant.conf on my sd card, i see somewhere that you can only create wpa_supplicant.conf on the first boot, that you cannot add wpa_supplicant.conf later to sd card

any help please.

klaus-ditze, 2019-09-05:
Think you can change it in the Web-Interface as admin in the Network Settings.

johnfc2018, 2019-09-13:
You can change /boot by remounting it read write:

mount -o remount,rw /boot

Then you can add a wpa_supplicant.conf file with network statements for each WiFi network you have. When you restart, it will use the wireless network it is configured to, so when you move it to a different area, it will find that one instead. 

For example:

country=gb
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
 scan_ssid=1
 ssid="MyNetworkSSID"
 psk="MyRouterKey"
}
network={
 scan_ssid=1
 ssid="MyOtherNetworkSSID"
 psk="MyRouterKey"
}

johannscholtz, 2019-09-13:
Great stuff, thx guys, it makes sense

joeydream, 2020-07-19:
do you mean after the setting, after doing somethings, we can move the  PRi zero motioneyeos system among  different ssid and without adding a wpa_supplicant.conf  file each time?

would you mind to say it more details, especially in the first step? i mean the mounting and remounting ? add a command of mount in which file ?

SupraGuy, 2020-08-26:
I don't know about yours, but mine, if it can't sign in to the network, it panics and reboots.

I set up my phone in access point mode with the same SSID and key as the old place, which I still was able to get, which allowed motioneyeos to boot and obtain an IP address. From there, I connected to the web interface, logged in as the admin account, and changed the wireless network from the Network settings tab.

This should also work if you can connect to a wired network.

juggledad, 2021-09-18:
I know this thread is old but I found that you need to change /data/etc/wpa_supplicant.conf