motioneye-project/motioneyeos #2094: Wyze Cam Pan RTSP Not Authorizing

chrisfish1, 2019-09-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Confirmed

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Confirmed

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry Pi 2 B+?

### Camera

I am using the following type of camera: Network Camera

My camera model is: Wyze Cam Pan with RTSP Firmware

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
### Log Files

I consider the following log files relevant to this issue:

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3586846/motioneye.log)
 * motioneye.log attached

I can view the RTSP stream through VLC just fine.  When I add it to Motioneye it just spins and never errors and never shows up.  The logs show what I believe to be an authentication issue.  I don't have another camera to test with but I am hopeful the fact that I can view it through VLC using RTSP shows that the camera is ready.
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3586849/motioneye.log)

JJRacoon, 2019-10-19:
I'm having the same issue with my Wyze.  The camera display shows "Unable to open video device" in the upper left corner of a blank screen.  VLC looks good using RTSP.

rcadmus, 2019-12-10:
Same problem.  motion only accepts the url if the user:pwd is left out.  Without the password it connects but can't authorize.  Any progress here?

rcadmus, 2019-12-11:
Solved...  Add the camera without the id:pwd  component because it won't accept the URL with it - thinks it is invalid.  Then edit /data/etc/camera#.conf adding the id:pwd to the url there. remove any trailing /   reboot the server for it to take effect. Works for me - hope this helps.

JJRacoon, 2019-12-14:
Solved... I found that the URL for Wyze network cameras should not 
include the username and password.  These should be input in the 
'Username' and 'Password' fields in the 'Add Camera' dialog.  When the 
'Camera' field shows recognition you should be all set.  I'm using 
Motioneye over 2019-09-26-raspbian-buster on my Pi 4 with 3 Wyze net 
cams.

<comment@noreply.github.com>
Sent: 12/11/2019 12:18:27 PM
(#2094)