motioneye-project/motioneyeos #2095: Wiki *Improvement "Controlling Camera LED using GPIO"  Board specific

cpapplefamily, 2019-09-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

**### * bonus suggestion**
You really need to read trough all the documentation between both the MotionEyeOS and MotionEye repositories. I feel information is spread out a bit to much between these. Many of the topic are related but have different information. Having one Wiki page for both using a Tabbed [common solution] [MotionEyeOS solution] [MotionEye *installed solution] would help eliminate jumping between the two.

### motionEyeOS Version

I am running motionEyeOS version: (motioneyeos-raspberrypi-20190904).

### Board Model
I am using the following board/model: (Pi Zero W).

### Camera

I am using the following type of camera: (MMAL,).

My camera model is: ([https://www.arducam.com/product/arducam-m12-night-vision-ir-cut-raspberry-pi-camera/](https://www.arducam.com/product/arducam-m12-night-vision-ir-cut-raspberry-pi-camera/)).

### Network Connection

My motionEyeOS unit is connected to the network via: (WiFi, ).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 NA

### Log Files

I consider the following log files relevant to this issue:

NA

As an intermediate or advance noob with linux I wanted to post something and maybe ask the developers to add a section to the Wiki about how to control the IR cut filter installed on some of the NoIR cameras like this one: [https://www.arducam.com/product/arducam-m12-night-vision-ir-cut-raspberry-pi-camera/](https://www.arducam.com/product/arducam-m12-night-vision-ir-cut-raspberry-pi-camera/
)
After working through getting an action button displayed and how to disable the camera LED using the motioneye wiki's the next step was to identify how to programicaly control the camera LED that is also used to turn on/off the IR filter.  Many web searches identify GPIO5 as the output to control for this purpose and some identify GPIO32 as the pin. Neither of these work. I was using the sample code for action buttons [https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons](https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons) for controlling a "Light" but changed the GPIO assignment. 

After confirming the data in the /sys/class/gpio/gpio$GPIO/value file was updating with 0 and 1 i visited multiple other web forums until I cam across [this post](https://raspberrypi.stackexchange.com/questions/70624/pi-zero-w-switching-camera-led-from-program-or-command-line)  calling out GPIO 40 

I can now control the Camera LED and simultaneously the Cut out filter. I assume the other GPIO pins references **are** for other makes of the Raspberry Pi. A wiki page of common code snippets or FAQ entry to address this could help many people.

Key points to include for this situation are to Disabling camera LED:
if running MotionEyeOS then you can use the motionEye web interface
If not using the MotionEyeOS but a manual install you will have to edit the  `/boot/config.txt `adding the line : `disable_camera_led=1` **BUT** remember you MUST first change the file system from protected to writable using 
```
mount -o remount,rw /
mount -o remount,rw /boot 
```

cpapplefamily, 2019-09-08:
My next adventure here will be to try and create something to schedule the IR cut filter on/off or use a photo sensor to control the IR  Cut out GPIO40 pin. The Camera IR lights already have a photo sensor to  turn them on and off but I'm not aware if I can get the status of the sensor in code. 

Either way I will have to solve how to run a back ground program. I seen references to this in my above adventures.
Maybe scheduler_1 addon like the monitor_1 Monitor Command could be useful? 
or
Maybe the Monitor Command I what I need? but checking the PhotoSensor every 1 sec is excessive. 
pseudo code:
```
PhotoSensor = GPIO17
IR_CutOut = GPIO40

if GPIO$PhotoSensor then
    IR_CutOut = 1
else
    IR_Cutout = 0
endif
```

jasaw, 2019-09-09:
Ideally, yes, the wiki should have this information to help people set up IR-cut filter, but it is not officially supported at the moment. We haven't thought about all the different scenarios of how this will be used, on which platform, etc. At a minimum, I know that controlling Pi Camera LED is different on different RPi models, and on RPi 3 and above, it requires a C program as the GPIO has moved from the CPU to GPU (I think... it's been a while since I looked into this).

This mechanical IR-cut filter thread may be useful to you: https://github.com/ccrisan/motioneyeos/issues/1013

I'm using this exact camera myself, but without the provided IR-LEDs with light sensor. The light sensors on the IR-LEDs are not accessible from GPIO (at least on mine), which is why I chose not to use them. I built my own light sensor with an [IR filter](https://www.ebay.com.au/itm/10pcs-Optical-UV-IR-Cut-Light-Filter-Blocking-For-Camera-Astronomy-Lens-8mm-Slim/282653207704) in front of it, otherwise it will be flooded with light when you turn on IR. Anyway, here's my camera hardware design if you are interested: https://github.com/jasaw/camera_hardware/tree/master/rpi-zero-w

Reading the light sensor is actually a bit complicated. To prevent erroneous reads and erratic IR-cut switching, I wrote a C program that debounces the raw reading and has programmable hysteresis thresholds. The source code and design are all here: https://github.com/jasaw/ldr-reader