motioneye-project/motioneyeos #2102: Camera not found after 20190427

duckietm, 2019-09-13:
After version 20190427 the Microsoft USB camera isn't found anymore :

/dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0
V4L2 Camera

If I install version 20190427 all works perfect but if I install all the later builds the camera isn't found in the GUI

ccrisan, 2019-09-14:
@duckietm please use the issue template to provide proper details about your problem. If we're talking about a Raspberry Pi, does it work on Raspbian?

duckietm, 2019-09-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427 

### Board Model

Raspberry PI 3B+

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

Microsoft_Microsoft®_LifeCam_HD-3000
V4L2 Camera

### Network Connection

Ethernet

duckietm, 2019-09-18:
Camera works fine on Raspbian and motioneyeOS 20190427 but all after that the camera isn't found anymore.
I have tryed it with 3 different camera's and RPI 3+ all the same result, the camera isn't found on version : 20190619 / 20190911

duckietm, 2019-09-19:
I found a fix.

- unplugged the camera and the rebooted.
- plugin the camera after the reboot, and all starts to work … ?

closed 2019-09-20 by duckietm