motioneye-project/motioneyeos #2103: Camera doesn´t work

Sisko4, 2019-09-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427

### Board Model

I am using the following board/model: Raspberry PI 3B

### Camera

I am using the following type of camera: MMAL

My camera model is: official cam v2.1

### Network Connection

My motionEyeOS unit is connected to the network via: WLAN

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

-----------------------------------------------------------

Hello!
I have a problem with my current setup of motioneye, the cam feed simply doesn´t show up.
I have a fresh install of motioneye, everything works but the cam!
I tried the cam on raspbian and there it works just fine.
Cam is okay, also the 3B.

Sisko4, 2019-09-14:
Okay, i think there is a problem with motion.
I just activated "fast network camera" and it suddenly works!
But i want to use the features like motion detection for timelapse pictures.

unk1nd, 2019-09-14:
Have the same issue on 4 of my pi Zero and tried with 6 of my camera modules that I know works.

`
[root@meye-f1cf0cc2 /]# lsmod
Module                  Size  Used by    Tainted: G
ipv6                  398081  0 [permanent]
spidev                  7220  0
brcmfmac              273262  0
brcmutil                9318  1 brcmfmac
snd_soc_bcm2835_i2s     7385  0
regmap_mmio             3874  1 snd_soc_bcm2835_i2s
snd_soc_core          178253  1 snd_soc_bcm2835_i2s
snd_bcm2835            23253  0
snd_compress            9956  1 snd_soc_core
snd_pcm_dmaengine       5370  1 snd_soc_core
cfg80211              542402  1 brcmfmac
rfkill                 21476  2 cfg80211
snd_pcm                89814  4 snd_soc_bcm2835_i2s,snd_soc_core,snd_bcm2835,snd_pcm_dmaengine
i2c_bcm2835             6465  0
snd_timer              22428  1 snd_pcm
snd                    60280  5 snd_soc_core,snd_bcm2835,snd_compress,snd_pcm,snd_timer
spi_bcm2835             7520  0
uio_pdrv_genirq         3718  0
uio                    10038  1 uio_pdrv_genirq
fixed                   3033  0
i2c_dev                 6706  0
i2c_bcm2708             5741  0
bcm2835_v4l2           43350  0
v4l2_common             5762  1 bcm2835_v4l2
videobuf2_vmalloc       5721  1 bcm2835_v4l2
videobuf2_memops        1802  1 videobuf2_vmalloc
videobuf2_v4l2         15667  1 bcm2835_v4l2
videobuf2_core         30872  2 bcm2835_v4l2,videobuf2_v4l2
videodev              172900  4 bcm2835_v4l2,v4l2_common,videobuf2_v4l2,videobuf2_core
media                  26672  1 videodev

[root@meye-f1cf0cc2 /]# ls /dev/video*
/dev/video0

[root@meye-f1cf0cc2 /]# v4l2-ctl --list-devices
mmal service 16.1 (platform:bcm2835-v4l2):
        /dev/video0
`

Clean install on all of MotionEyeOS.
GUI just gives me the spinning thingy.

By activating "fast network camera" , the image show's in GUI.
In my case this solved my issue after some hours due to having a Virtual Machine running MotionEye on Ubuntu that this Pi's are suppose to send the feed to.

Sisko4, 2019-09-14:
The curious thing is, i had this os a year ago on my ZeroW and it worked without any problems from the start. I hope that it is gonna be solved soon, because i need a live feed for my clients that order 3d prints, so they can watch the progress and i can get some timelapse material out of this.

unk1nd, 2019-09-14:
have you considered octopi (https://octoprint.org/) for that ?

unk1nd, 2019-09-14:
or try out the pre-release https://github.com/ccrisan/motioneyeos/releases/tag/20190911

Sisko4, 2019-09-14:
I tried octoprint a few times, but it doesn´t work very well for me, plus the octolapse feature doesn´t work with my printer and i have better results and control with the integrated system.
I will try out the pre-release or wait till a stable is released.

jasaw, 2019-09-15:
@Sisko4 I recommend trying the latest pre-release. There's an issue with Chrome that the latest pre-release has addressed. It might work for you.

auriux, 2019-09-16:
When the release version comes out?

Sisko4, 2019-09-17:
I tried the pre-release today, it starts up, connects to the wlan, but that is all it does.
No interface. I can see the ip and the hostname in the router, but i can´t connect to it.

dewexdewex, 2019-09-18:
I can't get the cam to display an image in the MotionEyeOS UI with my RPi Zero W setup, when using Chrome on Mac OSX. Safari is fine on MacOSX and Chrome is fine on iOS. I am using the latest build of the MotionEyeOS for the board.

Chrome Version 76.0.3809.132 (Official Build) (64-bit)

unk1nd, 2019-09-18:
Well, it is posted that the pre-release include the fix but the latest release does not

Sisko4, 2019-09-19:
Okay, i re-flashed the card with the pre-release again and now it works, must have been an error while flashing.

SureshkumarSenthilkumar, 2023-01-05:
Hii... I am using RaspberryPi 4 model B and isnatlled motioneyeOS latest version 
vcgetcmd showing no camera supported

starbasessd, 2023-01-05:
Use RPi4 dev20201026, not 20200606.

SureshkumarSenthilkumar, 2023-01-05:
> Use RPi4 dev20201026, not 20200606.

I installed that but it has only raspberry pi common os so it going rainbow screen I don't know what issue

SureshkumarSenthilkumar, 2023-01-06:
> There isn't a startx=1 in motionEyeOS. There is no desktop GUI environment.

Then how to enable the camera in cli