motioneye-project/motioneyeos #2106: Pi Zero W newer releases getting stuck, not working

puresilk, 2019-09-19:
I am a Raspberry Pi approved reseller, and sell kits including motioneyeOS (an older version)
I have tried several recent motionEyeOS releases with a very recent Pi Zero W (just arrived from the factory) in order to ship newer versions to customers. They all hang during boot with the same message "loading kernel modules", even if given ample time to finalize the loading. 
I have preconfigured all of these SD cards with a wpa-supplicant.conf before booting up motioneyeOS. 

ccrisan, I can test the newest version with a Pi Zero W, make sure it does not work, and can ship it (or several if you want) to you on my cost, as a contribution to your project. Please let me know if you would like that, to allow easier debugging for you.

### motionEyeOS Version

I have tested the following releases (all Pi 1 releases):

- 20190911 
- 20190904 ( I believe - am unsure at the moment whether I really tested that one, but I believe so)
- 20190427 

### Board Model

I am using the following board/model: Pi Zero W.

### Camera

Default Raspberry Pi camera. 

### Network Connection

WiFi 

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files

none were created on the root partition

puresilk, 2019-09-19:
BTW: I tried going through other Pi Zero W issues, most seem unrelated to this; as they fail to work at a later or earlier stage during boot.

YannickGro, 2019-09-19:
I'm sorry I cant help you with your problem. But I hope you can help me with a small question I have. What version should I use for the Pi Zero W? Its not clearly named bij the downloads. I'm looking the latest release version 20190427. 

Hope you can help!
[image]

ccrisan, 2019-09-19:
@puresilk could you try and make a short video recording of the issue? Recovering the log files from the SD card when that happens could also help.

ccrisan, 2019-09-19:
@YannickGro https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices will tell you to download the raspberrypi (1) version.

puresilk, 2019-09-20:
@ccrisan, I will make a video recording on the week-end. Where are the log files stored? I did not notice any on the FAT partition after removing the SD card.

ccrisan, 2019-09-20:
@puresilk they are on the large EXT4 partition, accessible from Linux.

avanc, 2019-09-27:
Don't know if it is related: Just started today with motioneyeos on a Raspberry Pi Zero. Booting worked as expected, by I get reboots every now and then (3 times within the last 30 minutes while playing around).

ccrisan, 2019-09-29:
@avanc version and logs please.

avanc, 2019-09-29:
Sure.

motionEye Version: 0.41
Motion Version: 4.2.2
OS Version: motionEyeOS 20190911

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3667294/dmesg.log)

Is the stack traces related to Wifi? I can connect via SSH to the device and I'm asked for the password. However, I don't get a prompt afterwards.

ccrisan, 2019-09-29:
The stack traces in your log are definitely related to WiFi. Try reducing the ARM frequency to 950 in your `/boot/config.txt`. It might be related to a problem identified with some Zero W boards.

avanc, 2019-09-30:
@ccrisan Thanks for the quick response. That works for me.
With this information I was also able to find the other issues related to my problem.
As it seems to be different than @puresilk's issue, I write down my results in https://github.com/ccrisan/motioneyeos/issues/1908.

puresilk, 2020-12-30:
Sorry for not following up on this earlier, "life got in the way". Here's a screenshot of the boot. It happens also when I unplug & reboot. I'm using the most recent release. 

[image]

I don't seem to be able to find a dmesg.log or any other log file on the SD card after removing it (after the second boot attempt). I assume that this log is never written to the SD card.

starbasessd, 2020-12-30:
On the back of the Pi, it says which model it is, can you check and see?
One of mine says:
Raspberry Pi Zero W V1.1
(C) Raspberry Pi 2016
Can you try dev20201026 or 20190911 please? Found here:
Releases · ccrisan/motioneyeos
<https://github.com/ccrisan/motioneyeos/releases>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

puresilk, 2021-01-04:
Possibly the Pi Zero W I'm using is defective; it works on another one. I'm closing this for now.

dhawker, 2021-11-01:
Pi Zero 2 W arrived the other day, works fine with Pi OS and with Octoprint but just freezes with MotioneyeOS,
any suggeswtions ( no logs saved, screen just sits at four colours)

puresilk, 2021-11-01:
As I've written on this page here:
[Everything about the Pi Zero 2 W](https://picockpit.com/raspberry-pi/everything-about-raspberry-pi-zero-2-w/)

Raspberry Pi Zero 2 W needs up-to-date boot files. Try fixing motionEye OS boot with copying the boot files as I've outlined, and report back whether that works.

starbasessd, 2021-11-01:
Could you please open a new ticket for this issue (with all the questions answered) for the PiZeroW2, your issue, and solution found, please, link it here, and close this issue? The PiZero/W/H is only related to the PiZeroW2 by form factor. The chips used are very different (both CPU and WiFi) and the issues don't apply to both.