motioneye-project/motioneyeos #2110: Rpi moved to a switch, motioneye now not showing camera streams

x24garcia, 2019-09-23:
I'm using version 20190427 on a Rpi3 with RTSP network cameras (iCamera 1 and 2), all 3 devices connected via wifi.

I've recently set up an old router as a switch and connected my current Rpi's to it using ethernet, including the one running Motioneyeos. Now the motioneye control panel can't display the camera streams and just keeps spinning.  The motion detection still works though, pictures and videos get saved, emails get sent. Just the streaming doesn't work.  Both cameras stream fine using IP Cam viewer basic though.

talormanda, 2019-09-30:
Update to 20190911. There is a chrome bug where it just displays the spinning icon that gets fixed in this release. This is apparent by using another browser and seeing that it works there instead.

x24garcia, 2019-09-30:
Checked with IE and it worked.  I've just updated to 20190911 and the spinny went away.  Thanks.

closed 2019-09-30 by x24garcia