motioneye-project/motioneyeos #2113: Growing video feed latency till lose connection

Destromite, 2019-09-28:
I'm having a very strange problem. When I view the feed directly from the camera using chrome its quite smooth and what I'd expect from the video footage. Very rarely jumping. But if I use Windows explorer or Firefox it is very laggy and choppy. if I open multiple feeds to the camera the input latency starts to get longer and longer growing more unstable until the camera loses signal with all devices and I have to restart it. Including if I have a separate hub for recording even if it's the only device talking to it. The footage is always choppy and laggy through the recording hub and doesn't load the camera streams from the hub on chrome so I have to go to firefox to even view it. And if I'm viewing the camera feed from the hub and open a tab to view it directly it won't load on firefox but will on chrome. And if I have the camera plugged into an HDMI port everything is smooth as butter but the remote view I still gets crappy till it loses signal. what's even stranger is that can leave the camera on for days and when I tune into it will be instantly responsive as it usually would be from the start but after viewing for just a few seconds if I come back to it I won't get a signal because of the growing latency problem. I assumed it was because of my 60$ Amazon router but I ordered a new different one and it still has the same problem. The camera is a pi 3 b+ in fast camera mode. Streaming to a pi 3 b+ for recording. I'm wanting to set up multiple of these but I have to get one working first. I have another Pi Z camera with the same issue so I not know it's not just the specific pie. Help would be appreciated. They are all running their respective version of: motion eyeos 20190427.

Destromite, 2019-10-15:
I was hoping I'd have some replies by now. Feels bad. I'm going to have to just mess with stuff till I find something that works I guess

jasaw, 2019-10-16:
@Destromite Sorry, the issue you described is indeed quite strange. Let's try to break this problem down...
1. Are you streaming over Ethernet or Wifi? If you're on Wifi, try switching over to Ethernet.
2. You mentioned Fast Camera Mode. Are you running RTSP or MJPEG protocol?
3. Try the latest nightly build. RTSP Fast Camera Mode has been reworked extensively.
4. Try lowering the video resolution and framerate.

Destromite, 2019-10-24:
Sorry for the late reply and thank you for trying to help. I have tried running it on low refresh rate and resolution to test if it was caused by the resources being overloaded. It didn't make a very noticeable difference. I'm using a fast camera mode on mjpeg. The cameras are over Wi-Fi to the router. The recording hub is ethernet connected to the router. The computer I've been using is connected to the router over Wi-Fi. I'll check again to see if there's a new update but last I checked it was up to date. Looks like there is an update hopefully it fixes this.

Destromite, 2019-10-24:
I believe I've found this to be a two fold problem. Ive connected the router directly to my computer and the camera. The feed is more responsive and doesn't grow in latency (at least not fast enough to notice in a few hours) but it seems to still have stutters and freezes. Lowering the fps makes it happen less and more subtle but its still there. The wifi looks to have a lot of other signal in the area and I've noticed it on using the computer sometimes. So i understand the feed being extra choppy or losing signal for a second or but it doesn't explain why it doesn't resync to real time after or why the device simply quits receiving input if it happens for too long from any device. I was also trying to switch it on to 5G band to bypass all the interference but the software doesn't seem to support it.