motioneye-project/motioneyeos #2114: Upstream Motion Revisions to Use Autopoint

Mr-DaveDev, 2019-09-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.  
**_Yes_**

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.  
**_Yes_**

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

**_Master branch as of 20190928_**

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
**_None.  Tested build of Rasperry Pi 3_**

### Camera
I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
**_None:  Tested build only._**

My camera model is: (insert your camera model, e.g. Logitech C290).
**_None:  Tested build Only_**

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
**_None:  Tested build Only_**

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
**_None:  Tested build Only_**

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

The current upstream master branch of Motion implemented Autopoint as part of a revision required from Gentoo distribution.  The inclusion and use of autopoint breaks the Buildroot upon which Motioneyeos is based.  

The following are a couple of options for your consideration

Option 1:  Eliminate NLS support from the upstream. [gist 1](https://gist.github.com/Mr-DaveDev/8363b732ef57aff1caee81decbc66c86) and [gist 2](https://gist.github.com/Mr-DaveDev/4d213af40bb4786c3a51535bf7cc1fdd) illustrate the required changes.

Option 2:  Implement hack to Buildroot methods to manually invoke autopoint and autoreconf.  [gist 3](https://gist.github.com/Mr-DaveDev/904c05ad12a05432a30510ad33f4661a) and [gist 4](https://gist.github.com/Mr-DaveDev/1b4c2b65f1a84fbc107e2b35c9e102aa) illustrate the required revisions.

Additional options considered included ` LIBFOO_GETTEXTIZE` which was inappropriate since it changes (and breaks) the upstream code.  Also upstream is considering whether it is now required to include and maintain all the m4 macros or whether NLS should just be depreciated.

ccrisan, 2019-09-29:
@Mr-DaveDev thanks for letting us know. I personally prefer Option 1 since motionEye doesn't really support any kind of I18N.

Do you see any problem if we disable NLS already for current builds based on Motion-4.2.2?

popoviciri, 2019-09-29:
Just gave option 1 a go in latest motioneyeos and after fixing a `'for' loop initial declarations` issue in mmalcam.c (very simple change here https://github.com/Motion-Project/motion/pull/1027), it compiles well.
Tested on a raspberry pi zero w with a pinoir v2 camera.
Cheers!

ccrisan, 2019-09-29:
@popoviciri can you please prepare a PR with your work, against our latest `dev`? The next release will probably come with the next version of Motion, so I believe it's safe to have it in dev already.

popoviciri, 2019-09-29:
Hi @ccrisan I made a PR (https://github.com/ccrisan/motioneyeos/pull/2117) but I feel it's missing a ton of adjustments as you can see in https://github.com/jasaw/motioneyeos/tree/xu4-hw-h264-new-motion branch. I recommend @jasaw have a look at it. Cheers!

closed 2019-10-21 by Mr-DaveDev