motioneye-project/motioneyeos #2116: Virual esxi

pergolafabio, 2019-09-29:
Hi, i have a USB camera, i only got ESXI servers, can i run motioneyeOS on a esxi server ? or is it only form arm processors?

jasaw, 2019-10-10:
https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices
Only the above devices are supported.

closed 2019-10-10 by jasaw

MarkProminic, 2020-10-31:
Sorry to bring up a closed issue, but now that ESXI supports Arm, I thought we could revisit this:

https://flings.vmware.com/esxi-arm-edition

I have tried to use instructions similar to the guide for Raspberry Pi OS on ESXI for Raspberry Pi:

https://www.virtuallyghetto.com/2020/10/how-to-run-raspberry-pi-os-as-a-vm-on-esxi-arm.html

Though I found that I couldn't install the grub packages as I am unfamiliar with how MotionEyeOS is built.

starbasessd, 2020-10-31:
Still doesn't work. the VM drives do not show as mmcblk0p1, 2,&3, and there
is a lot hard-coded to those devices.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-10-31:
If you want to sandbox it, you can try running it in a docker...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?