motioneye-project/motioneyeos #2118: *.svg's not loading

LucaMH, 2019-09-29:
Hi,

i am running motionEye server 0.41 on CentOS 7.7.1908 in a VM on an ESXi 6.7.0.

So far everything is running fine and smooth with 1 Cam (more to come) but the WebUI of the server isn't showing the SVGs for settings, swith user, etc...

I checked and they are in /usr/lib/python2.7/site-packages/motioneye/static/img but the webserver isn't loading them.
I checked the permissions on the files and they are fine (all on read) which doesn't wonders me, because the gifs from there are showing.
I then checked Sources in Chromes F12 menu and they aren't showing there
At last i checked the Pi which has the cam attached to it and there the icons are loading and sitting in the same path..

Has anyone ever had this before?

Thanks!
Luca

ccrisan, 2019-09-30:
Make sure you have the file `/etc/mime.types` present and that it has an entry for SVG.

LucaMH, 2019-09-30:
Hi,

file wasn't there.
I copied it over from the pi, rebooted the vm and refreshed the page without Cache.
Now the Icons are showing.

Thanks! :)

closed 2019-09-30 by LucaMH