motioneye-project/motioneyeos #2124: contant rebooting after install

pkelleratwork, 2019-10-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20190911

### Board Model

I am using the following board/model:pi zero w

### Camera

I am using the following type of camera: mini camera for pi zero w

My camera model is: JT-ZERO-V2.0

### Network Connection

My motionEyeOS unit is connected to the network via: wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

none

### Log Files
unable to get logs because os reboots

the os starts correctly, can connect to it via web broswer and ip address. after running for about 45 seconds, the os crashes and starts rebooting. on the second reboot the wap_supllicant file does not work and cannot get a connection.

jasaw, 2019-10-04:
I don't see anything obviously wrong in the logs. Maybe you're hitting the issue with new generation of Pi Zero (See https://github.com/ccrisan/motioneyeos/issues/1945)? Try downclocking to 950MHz.

pkelleratwork, 2019-10-04:
thanks - i'll give that a shot and post the results.

pkelleratwork, 2019-10-05:
@jasaw - i found this and tried it - seems to work now... https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=212777
TL;DR - added "over_voltage=4" to config.txt file 
thanks

closed 2019-10-05 by pkelleratwork