motioneye-project/motioneyeos #2127: Motion Capture Far to sensative

ukmercanary, 2019-10-06:
Does anyone else find that the motion capture is far to sensative? 
I just get an endless amount of uploaded photos of nothing to my Google Drive.
Using a Raspberyy Pi3 amd an old Foscam and a an IGeek camera.

gb-dursley, 2019-10-07:
If you login to motionEyeOS as admin and open the Settings (top left-hand corner of window), then go down to Motion Detection and open that sub-menu, you'll find a number of settings that allow you to tweak the motion detection sensitivity in a number of different ways.  From experience with this and other CCTV software, the only way to find the best settings for a particular purpose/location is to experiment - once you've had an initial stab at the settings, only change one setting at a time and not by a lot at once.  I find it helps to turn on Show Frame Changes (a red outline) so you can see what was the trigger and when.  Hope that helps.

gb-dursley, 2019-10-07:
Sorry, missed a crucial stage in my previous post:-  After opening Settings, under General Settings turn ON Advanced Settings, (only) then (can you) go down to Motion Detection...

ukmercanary, 2019-10-11:
Thanks very much checking now sorry for late reply