motioneye-project/motioneyeos #2129: motionEyeOS 20190911 unstable on Raspberry Pi zero W

Bra1nK, 2019-10-06:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911 

### Board Model

I am using the following board/model: Raspberry Pi zero W

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: vc.ril.camera.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that may/may not be relevant to this issue:

RTC DS1307

### Log Files

I consider the following log files relevant to this issue:

[0:vc.ril.camera] [ERR] [VID] camera_control_callback: Received unexpected camera control callback event, 0x4f525245
[0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout. Trying to do a graceful restart
[0:motion] [ERR] [ALL] motion_watchdog: Thread 1 - Watchdog timeout did NOT restart, killing it!

Web UI and then SSH become unresponsive frequently since upgrading to motionEyeOS 20190911 necessitating a power off/on.
After reverting to motionEyeOS 20190911 I'm back to stable operation and working round the lazy loading with chrome bug.

My RPI3 is working well after updating to 20190911  it's just the Zero W having problems

GrueMaster, 2019-11-29:
Similar issues here.  Same HW, same sw release.  I also am seeing the following kernel core dump in dmesg:
[  191.533598] Unable to handle kernel NULL pointer dereference at virtual address 00000074
[  191.533636] pgd = ecd5b58d
[  191.533648] [00000074] *pgd=15006831, *pte=00000000, *ppte=00000000
[  191.533676] Internal error: Oops: 817 [#1] ARM
[  191.536308] Modules linked in: ipv6 spidev brcmfmac brcmutil sha256_generic snd_soc_bcm2835_i2s regmap_mmio raspberrypi_hwmon hwmon cfg80211 snd_soc_core snd_bcm2835(C) snd_compress snd_pcm_dmaengine i2c_bcm2835 rfkill snd_pcm snd_timer snd spi_bcm2835 bcm2835_codec(C) v4l2_mem2mem videobuf2_dma_contig fixed uio_pdrv_genirq uio i2c_dev i2c_bcm2708 bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media
[  191.552523] CPU: 0 PID: 1334 Comm: st0 Tainted: G         C        4.19.65 #1
[  191.554536] Hardware name: BCM2835
[  191.556551] PC is at __sys_sendto+0x64/0x114
[  191.558600] LR is at 0xa
[  191.560551] pc : [<c058c868>]    lr : [<0000000a>]    psr: a0000013
[  191.562592] sp : d493be98  ip : c0729c00  fp : d493bf8c
[  191.564679] r10: 00000000  r9 : d493a000  r8 : 00004000
[  191.566777] r7 : 00000000  r6 : 0000000a  r5 : c09ef028  r4 : 00000000
[  191.568927] r3 : d5e65e61  r2 : 00000001  r1 : d5e65e60  r0 : d5689180
[  191.571109] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  191.573345] Control: 00c5387d  Table: 15d10008  DAC: 00000055
[  191.575569] Process st0 (pid: 1334, stack limit = 0xcd9aa47d)
[  191.577759] Stack: (0xd493be98 to 0xd493c000)
[  191.579937] be80:                                                       d493bebc c0a05bb8
[  191.584088] bea0: c004af68 fffffff7 00000001 b260b568 00000400 00000000 00000000 00000001
[  191.588167] bec0: 00000000 00000400 d493beac 00000001 d493bf14 00000000 00000000 00004040
[  191.592226] bee0: d527e200 c09f8448 633621e4 c09ef028 d493bf14 d527e200 d4928e00 c09f8448
[  191.596264] bf00: c09ef028 d52f3d40 d493bf74 d493bf18 c06d514c c004af18 c0a78694 d6cb84db
[  191.600537] bf20: 00000001 00000000 00000001 c06d5624 978e2bf0 d6cb84db 00000001 c09ef028
[  191.605054] bf40: 00000001 aa23cca0 000000a8 c00091a4 d493a000 00000000 d493bfa4 d493bf68
[  191.609718] bf60: c018af14 d6cb84db 00000400 00000400 00eadb28 156f61e1 00000121 c00091a4
[  191.614624] bf80: d493bfa4 d493bf90 c058c974 c058c810 00000000 00000000 00000000 d493bfa8
[  191.619672] bfa0: c0009000 c058c958 00000400 00eadb28 0000000a b260b568 00000400 00004000
[  191.624870] bfc0: 00000400 00eadb28 156f61e1 00000121 b260b500 aa23cca0 00000000 ffffffff
[  191.630231] bfe0: 00000000 aa23cc18 aa23cbf4 b5843154 80000010 0000000a 00000000 00000000
[  191.635748] [<c058c868>] (__sys_sendto) from [<c058c974>] (sys_send+0x28/0x30)
[  191.638595] [<c058c974>] (sys_send) from [<c0009000>] (ret_fast_syscall+0x0/0x28)
[  191.643987] Exception stack(0xd493bfa8 to 0xd493bff0)
[  191.646690] bfa0:                   00000400 00eadb28 0000000a b260b568 00000400 00004000
[  191.652001] bfc0: 00000400 00eadb28 156f61e1 00000121 b260b500 aa23cca0 00000000 ffffffff
[  191.657305] bfe0: 00000000 aa23cc18 aa23cbf4 b5843154
[  191.659920] Code: e1a00006 e24b10e8 e24b20e4 ebfff9f3 (e2506000) 
[  191.662697] ---[ end trace 9ffd4a10a18e38d1 ]---

justincardoza, 2020-02-09:
I'm having similar issues with the same hardware as well. I have a Pi Zero W with a Raspberry Pi camera v1.3 (type: MMAL Camera, device: vc.ril.camera) running motionEyeOS 20190911. I've noticed that there seems to be gradual system degradation over time: first the camera will lose connection but the web interface still works, and sometimes changing a camera setting will bring back the image. Then after a while the web interface goes down but I can still SSH in. Then SSH goes down but it's still pingable. Then eventually I get "destination unreachable" errors from ping attempts. A hard reboot usually fixes it for a day or so, but sometimes I have to cycle power multiple times to get it working again. I'm seeing these messages at the end of dmesg.log:

[  122.210439] Alignment trap: not handling instruction e1921f9f at [<c01410d0>]
[  122.210460] Unhandled fault: alignment exception (0x001) at 0x2bf4103a
[  122.210479] pgd = 69e0904b
[  122.210484] [2bf4103a] *pgd=00000000
[  122.210507] Internal error: : 1 [#1] ARM
[  122.210734] Modules linked in: ipv6 spidev brcmfmac brcmutil sha256_generic snd_soc_bcm2835_i2s regmap_mmio snd_soc_core cfg80211 raspberrypi_hwmon hwmon snd_bcm2835(C) snd_compress snd_pcm_dmaengine rfkill snd_pcm i2c_bcm2835 snd_timer snd spi_bcm2835 bcm2835_codec(C) v4l2_mem2mem videobuf2_dma_contig uio_pdrv_genirq uio fixed i2c_dev i2c_bcm2708 bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media
[  122.212113] CPU: 0 PID: 1369 Comm: grep Tainted: G         C        4.19.65 #1
[  122.212338] Hardware name: BCM2835
[  122.212483] PC is at do_wp_page+0x94/0x514
[  122.212626] LR is at unlock_page_memcg+0x1c/0x20
[  122.212768] pc : [<c01410d4>]    lr : [<c016b9e0>]    psr: a0000113
[  122.212932] sp : d4ef9e00  ip : d4ef9d98  fp : d4ef9e2c
[  122.213070] r10: 00000055  r9 : d7018800  r8 : 00000040
[  122.213241] r7 : 00000055  r6 : d4ee7840  r5 : c09ef228  r4 : d4ef9c3c
[  122.213447] r3 : 2bf4103a  r2 : 2bf4103a  r1 : 00000000  r0 : 00000008
[  122.213660] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  122.213850] Control: 00c5387d  Table: 16bb4008  DAC: 00000055
[  122.214028] Process grep (pid: 1369, stack limit = 0x3a2e0da1)
[  122.214185] Stack: (0xd4ef9e00 to 0xd4efa000)
[  122.214298] 9e00: d4ef9e2c d4ef9e10 c00d4044 2bf4103a b6f5a000 d4ee7840 c09ef028 00000055
[  122.214483] 9e20: d4ef9e9c d4ef9e30 c0142664 c014104c 00000000 d4ef9e40 d7018800 d4ee7840
[  122.214665] 9e40: 00000055 006200ca 00000129 b6f5a000 d6bb6db8 d6bb6db8 00b8f38f 00000000
[  122.214845] 9e60: 00000000 d7ca3c1c d4941568 d6a24858 00000000 2bf4103a d4ef9fb0 d4c0e200
[  122.215026] 9e80: 0000081f b6f5a6c4 d6a24820 d6a24858 d4ef9ef4 d4ef9ea0 c06da278 c014215c
[  122.215220] 9ea0: b6fc3000 00000022 00000003 00000000 00000073 d6a24820 d4ef9f04 00000000
[  122.215401] 9ec0: 00000800 00000000 000b6fc3 c09f4580 0000081f c09ef028 c06d9f54 b6f5a6c4
[  122.221640] 9ee0: d4ef9fb0 b6fc4928 d4ef9fac d4ef9ef8 c001ab44 c06d9f60 d4ef9f5c d4ef9f08
[  122.228023] 9f00: c012d59c 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  122.234038] 9f20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  122.240367] 9f40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  122.246619] 9f60: 00000000 00000000 b6fa354c c0009c7c ffffffff 00c5387d 00c5387d b6fc5d90
[  122.252635] 9f80: d4ef9fac 2bf4103a b6fa354c 20000010 ffffffff 00c5387d 00c5387d b6fc5d90
[  122.259392] 9fa0: 00000000 d4ef9fb0 c0009c84 c001ab00 b6e21000 001396c4 b6f5cc48 b6e3434c
[  122.266243] 9fc0: b6e3689c b6e36b3c 00000001 b6fc6000 b6e21000 b6fc5d90 b6fc4928 bea86ad4
[  122.272796] 9fe0: b6e21000 bea86a10 b6fc6560 b6fa354c 20000010 ffffffff 00000000 00000000
[  122.280764] [<c01410d4>] (do_wp_page) from [<c0142664>] (handle_mm_fault+0x514/0xc68)
[  122.287237] [<c0142664>] (handle_mm_fault) from [<c06da278>] (do_page_fault+0x324/0x3b8)
[  122.294037] [<c06da278>] (do_page_fault) from [<c001ab44>] (do_DataAbort+0x50/0xf4)
[  122.300584] [<c001ab44>] (do_DataAbort) from [<c0009c84>] (__dabt_usr+0x44/0x60)
[  122.307443] Exception stack(0xd4ef9fb0 to 0xd4ef9ff8)
[  122.314276] 9fa0:                                     b6e21000 001396c4 b6f5cc48 b6e3434c
[  122.321066] 9fc0: b6e3689c b6e36b3c 00000001 b6fc6000 b6e21000 b6fc5d90 b6fc4928 bea86ad4
[  122.328504] 9fe0: b6e21000 bea86a10 b6fc6560 b6fa354c 20000010 ffffffff
[  122.336038] Code: 12423001 e283201c f5d2f000 e1921f9f (e2811001) 
[  122.344487] ---[ end trace cf7cb924de65ae4b ]---