motioneye-project/motioneyeos #2133: os.conf does not get used on first boot

jasperza, 2019-10-07:
I created a new image of motioneye os on an sd card (using windows 10). it all works great ! 

Then I wanted to turn off networking (OS_NETWORKLESS="true" to run it as an AP), so I created a new SD with a clean image and  put a modified os.config file (attached) into the root folder of the SD card before first boot.

I booted it into the rPi4 and it ignored the os.config file on first boot and left it in the root folder  of the sd card. If continued with networking enabled

There is no easy way to modify the os.config file afterwards because it needs to be inserted into another linux/RPI machine , but there are no permissions when editing the file, i need the os.config file to be recognised and used on first boot like the other files (dnsmasq.conf and hostapd.conf ) which are recognised on first boot and copied off the root folder of the sd card sucessfully, no further editing needed.

I am running motionEyeOS version: motioneyeos-raspberrypi4-20190911.img
I am using the following board/model: Raspberry PI 4 B 2GB
I am using the following type of camera:  Pi Cam b.

popoviciri, 2019-10-07:
Your file's End Of Line (EOL) is Windows (CR LF) formatted. You need to convert it to Unix (LF) before placing onto BOOT partition. You can use notepad++ to convert: Edit -> EOL conversion -> Unix.
Also, you don't need to add all the defaults in that file. Only what you want to change. Cheers!

jasperza, 2019-10-08:
> Your file's End Of Line (EOL) is Windows (CR LF) formatted. You need to convert it to Unix (LF) before placing onto BOOT partition. You can use notepad++ to convert: Edit -> EOL conversion -> Unix.
> Also, you don't need to add all the defaults in that file. Only what you want to change. Cheers!

popoviciri, you are a star! thanks so much for this tip, it worked! I guess I was thrown because the other files worked and I edited those in normal notepad. thanks so much for taking the time to respond.

closed 2019-10-08 by jasperza

ccrisan, 2019-10-08:
There's a preprocessing mechanism in place that does that transformation automatically for various config files. It seems that it doesn't work for `os.conf` - I'll have to verify this.