motioneye-project/motioneyeos #2135: SNTP not being used

popoviciri, 2019-10-07:
In `S50date` startup script: calling `source ${CONF}` after `DATE_METHOD=sntp` overrides the latter to whatever is set in the frontend: `ntp` or `http`. One solution would be to rename `set_current_date_sntp` to `set_current_date_ntp` and scrap the `ntpq -gq` option entirely.
There are many other ways to sort this, but if the one described here is acceptable, I can submit a PR.
By the way: SNTP is a blitz in comparison with NTP. Good change!

ccrisan, 2019-10-08:
@popoviciri you're right: motionEye's UI doesn't know about SNTP.

I believe the least invasive and backwards compatible way is to teach motionEye about SNTP, while leaving the old NTP in place so that users who prefer NTP's precision over SNTP's speed will be happy to.

ccrisan, 2019-10-08:
https://github.com/ccrisan/motioneyeos/pull/2137 is my proposed fix for this issue.

popoviciri, 2019-10-08:
Hi @ccrisan, I quickly tested this and `date.conf` is being updated accordingly. Made a comment on the PR. Cheers!

closed 2019-10-08 by popoviciri