motioneye-project/motioneyeos #2140: Unable to use FNC RTSP with 2592x1944 resolution

brwainer, 2019-10-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: 
 * Board A: Raspberry Pi 4B 4GB
 * Board B: Raspberry Pi 4B 4GB
 * Board C: Raspberry Pi 3B+

### Camera

I am using the following type of camera: 
Fast Network Camera with RTSP Streaming

My camera model is: 
 * Board A: Official Raspberry Pi Camera V2.1 (8MP)
 * Board B: Generic 5MP OV5647 NO-IR Camera (https://www.amazon.com/dp/B07HMH1C59/)
 * Board C: Official Raspberry Pi Camera V2.1 (8MP)

### Network Connection

My motionEyeOS unit is connected to the network via: 
Both Wired and Wireless (unrelated to this issue)

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

 * messages.log prints out "user.notice streameye: not running, respawning" every 5 seconds
 * the rest of the logs from Expert Settings do not contain relevant information

### Issue Summary

On three different boards, RTSP streaming works at most resolutions, but fails at 2592x1944. 1640x1232 and 1920x1080 both work. I have tried changing the GPU memory up to the maximum of 944MB (On one of the Pi 4B 4GB) and there was no change in behavior. On the 3B+ (the only board that shows me a local image for any resolution while in RTSP mode, a separate issue will be submitted for that), the image shows for all of "working" resolutions, but not for 2592x1944. When viewing the streams with VLC, when any of the boards is set to 2592x1944, VLC reports "Your input can't be opened: VLC is unable to open the MRL 'rtsp://192.168.10.91:554/h264'. Check the log for details." and the logs show that it can connect to the RTSP server, but there is no video feed available.

brwainer, 2019-10-10:
One other symptom: after setting a board to 2592x1944 with RTSP, the RTSP stream will still fail after changing the resolution back down to something lower. The board must be rebooted for RTSP to work at the lower resolution again.

jasaw, 2019-10-10:
RTSP streaming won't work at resolutions above 1920x1080. That's the max resolution the hardware H264 encoder can handle. The bug here is higher resolutions shouldn't be selectable from the web front-end.

jasaw, 2019-10-10:
Once you've gotten the GPU into an error state, there's no way to recover it except for a reboot. Pushing higher than 1920x1080 resolution to the GPU will get it into an error state.

jasaw, 2019-10-14:
@ccrisan any good idea on how to limit the resolution based on whether RTSP or MJPEG is selected? I imagine the web front end will need to automatically lower the resolution if it's higher than 1920x1080 for RTSP mode.

jasaw, 2019-10-16:
@ccrisan I think a help text and some code in `streameyectl.py` to automatically limit the resolution to highest supported while maintaining the aspect ratio.

ccrisan, 2019-10-16:
@jasaw yes, hard limiting it from streameyectl.py should be doable as well.

jasaw, 2019-10-30:
Should be fixed in the latest nightly build.

closed 2019-10-30 by jasaw