motioneye-project/motioneyeos #2141: Image Preview of FNC RTSP Does not work on Raspberry Pi 4B (works on 3B+)

brwainer, 2019-10-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: 
 * Board A: Raspberry Pi 4B 4GB
 * Board B: Raspberry Pi 4B 4GB
 * Board C: Raspberry Pi 3B+

### Camera

I am using the following type of camera: 
Fast Network Camera with RTSP Streaming

My camera model is: 
 * Board A: Official Raspberry Pi Camera V2.1 (8MP)
 * Board B: Generic 5MP OV5647 NO-IR Camera (https://www.amazon.com/dp/B07HMH1C59/)
 * Board C: Official Raspberry Pi Camera V2.1 (8MP)

### Network Connection

My motionEyeOS unit is connected to the network via: 
Both Wired and Wireless (unrelated to this issue)

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

N/A

### Issue Summary

On the Pi 3B+ when using FNC with RTSP Streaming at a working resolution (separate issue submitted for 2592x1944 not working), the "local" webpage shows the image preview. On both of the Pi 4B 4GB, with identical settings to the Pi 3B+, the "local" webpage does not show the preview image, it just tries to load indefinitely. Meanwhile, streaming with VLC works fine on all three boards.

jasaw, 2019-10-10:
The FNC RTSP has just been reworked extensively in the latest nightly build. You could give it a whirl if you want.
https://github.com/ccrisan/motioneyeos/releases/tag/nightly-dev

brwainer, 2019-10-10:
This issue is resolved in dev20191010. Thank you! Should I close this issue now, or wait to confirm that the next stable release has it resolved also?

jasaw, 2019-10-10:
Let's close the issue.

closed 2019-10-10 by jasaw