motioneye-project/motioneyeos #2142: AWB "Greyword" does not work in FNC RTSP mode

brwainer, 2019-10-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: 
 * Board A: Raspberry Pi 4B 4GB
 * Board B: Raspberry Pi 4B 4GB
 * Board C: Raspberry Pi 3B+

### Camera

I am using the following type of camera: 
Fast Network Camera with RTSP Streaming

My camera model is: 
 * Board A: Official Raspberry Pi Camera V2.1 (8MP)
 * Board B: Generic 5MP OV5647 NO-IR Camera (https://www.amazon.com/dp/B07HMH1C59/)
 * Board C: Official Raspberry Pi Camera V2.1 (8MP)

### Network Connection

My motionEyeOS unit is connected to the network via: 
Both Wired and Wireless (unrelated to this issue)

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

N/A

### Issue Summary

When I change from MJPEG streaming to RTSP streaming, the "Greyworld" AWB no longer is applied (on Board B which has a NOIR camera), the image looks like the AWB goes back to Auto or Off. Setting Tungsten also has no effect - I assume this affects all AWB modes, but I have not tested them.

jasaw, 2019-10-10:
Try the latest nightly dev.

brwainer, 2019-10-10:
This issue is resolved in dev20191010. Thank you! Should I close this issue now, or wait to confirm that the next stable release has it resolved also?

jasaw, 2019-10-10:
Let's close the issue now. Thank you for testing. :-)

closed 2019-10-10 by jasaw