motioneye-project/motioneyeos #2144: Cannot send Images thru GMAIL (Error: Check SMTP server name!

vezcu2000, 2019-10-11:
Cannot get past this issue.
I have tried less-secure account and normal account

Email Adresses: my-gmail-address@gmail.com
SMTP Server: smtp.google.com
SMTP Port: 587
SMTP Account: my-gmail-address@gmail.com
SMTP Password: my-gmail-password
From Adress:  my-gmail-address@gmail.com
Use TLS: ON
Attached Pictures Time Span: 10

Got error when pressing a button Test Email

I have tried also setting port 465 and TLS OFF..

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911.

### Board Model

I am using the following board/model: Raspberry PI 3B+.

### Camera

I am using the following type of camera: PI NOIR V2

My camera model is: PI NOIR V2

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi or Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

USB bluetooth adapter Rapoo for Keyb and mouse

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://www.dropbox.com/s/dmkm5jnwfk0912h/boot.log?dl=0)
[motioneye.log](https://www.dropbox.com/s/dpzgp2uwfxp3ut0/motioneye.log?dl=0)

Dewey3, 2019-10-12:
This is a shot in the dark, but try changing you SMTP address to smtp.gmail.com, and your SMTP port to 465.

Dewey3, 2019-10-12:
Also, for the SMTP account, try dropping the "@gmail.com"

vezcu2000, 2019-10-14:
Hello.

> This is a shot in the dark, but try changing you SMTP address to smtp.gmail.com, and your SMTP port to 465.

It worked for this problem.

However.. now it says "Notification email failed: Connection unexpectedly closed: timed out!

Motioneye log says

`2019-10-14 09:45:59: [motioneye]    ERROR: notification email test failed: Connection unexpectedly closed: timed out
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 827, in test
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 53, in send_mail
  File "usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "usr/lib/python2.7/smtplib.py", line 318, in connect
  File "usr/lib/python2.7/smtplib.py", line 366, in getreply
SMTPServerDisconnected: Connection unexpectedly closed: timed out
2019-10-14 09:45:59: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
`
I have tried with two different PI3 (using same pi noir v2 camera).
Wonder what I should try next. I also tried to google this problem, but
didn't get any clue for my case.

Thanks

vezcu2000, 2019-10-14:
Flashing motioneye-os to a new SD card and using latest PI3 image.

[picture](https://www.dropbox.com/s/28ajw7ff63zrckb/N%C3%A4ytt%C3%B6kuva%202019-10-14%20kello%2014.36.29.png?dl=0)

It does send the email but no pictures:
[settings](https://www.dropbox.com/s/ihggsug2wwl76gp/N%C3%A4ytt%C3%B6kuva%202019-10-14%20kello%2014.41.59.png?dl=0)

Dewey3, 2019-10-15:
Based on your pictures, I see where you dropped the "@gmail" from the Gmail SMTP account name, but did you change the port back to 465 when you tried (the picture shows 587)?  I would also have to reboot an older version of MotioneyeOS to get it to work once I got the parameter right.

vezcu2000, 2019-10-15:
Now when I changed back to port 587  and ckecked the TLS tap its working. I will get the pictures as attachment to my mailbox.

However I need to check how far it can detect motion using pi noir camera with 48 IR led array. It seems that after 3 meters it barely detects motion. Maybe PIR sensor would work better.