motioneye-project/motioneyeos #2146: Add Camera (20190911) as Network Camera => green screen, connection lost

Klausoleum, 2019-10-13:
I have the following structure:

**3 x Raspberry 3B as network camera**
two with the latest release 20190911 and one with 20190119

**1 x odroid hc1 as camera server** 
with latest release (20190911)

Unfortunately the stream is not displayed in the server when adding a network camera (Raspberrys with the latest release (20190911)).
After adding, the stream is shown in green. After a short while, a gray screen appears with "connection to camera lost".

As "MPEG Camera" and "Remote motioneye camera" I can display both streams in the server very well.

When adding the network camera with earlier release 20190119 there are no problems with adding and displaying in the server.

A change of the motioneyeos version (on 20190119) in the server did not bring any change.

It looks to me like there would be problems only when posing as a network camera with the latest release.

Is there any experience or knowledge regarding the latest release?

ccrisan, 2019-10-13:
I am using 20190911 with my Odroid XU4 and 9 netcams without any problem. Are your cameras RTSP or MJPEG netcams? Can you provide some logs?

Klausoleum, 2019-10-13:
I configured all cameras as normal network cameras with motioneyeos (no fast network camera) with MPEG Stream (ME OS does not support RTSP or?)

But I just found out that:
When I change the cameras to "fast network camera" mode, there are no problems to display the stream in the server as a network camera.
After two tries to aktivate the Cams on RPi as a fast network camera and disable it I could even add the normal network camera without errors in the server.

Currently I do not want to risk another game, I'm happy that it works so first.

There are some errors in the log fils, but they do not indicate the cause for me.

[messages.log](https://github.com/ccrisan/motioneyeos/files/3722153/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3722154/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3722155/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3722156/boot.log)
[image]

Klausoleum, 2019-10-13:
I have some news to reproduce the error:

If the framerate in the server is greater than or equal to the framerate of the network camera (the set, not the transmitted) a green screen and after a short time "connection to camera lost".

the attitude as described above, I think logically, but the error should be intercepted by software or?