motioneye-project/motioneyeos #2147: Critical error on a brand new setup afyt

jeanfbrito, 2019-10-14:
Starting motioneye: CRITICAL:Root:failed to read settings from "/data/etc/motion.conf": invalid configuration line: motion_binary

ccrisan, 2019-10-15:
`motion.conf` is used by `Motion` and is automa ically created/managed/updated by motionEye, while `motioneye.conf` is used by motionEye as a config file. The directive you mentioned belongs to motionEye, so you must have mixed up the config files somehow.

jeanfbrito, 2019-10-15:
It's a brand new instalation with the `ready image`. I hadnt done nothing. Just installed the image on the SDcard, and turn on the raspberry. Then, by wire network I have access to it, configure wifi, it reboots and never works again.

jeanfbrito, 2019-10-15:
`motioneyeos-raspberrypi3-20190911.img`

jeanfbrito, 2019-10-15:
The message is what shows on the screen connected to the HDMI on the boot after wifi configured. Nothing works and it reboots again and again.

ccrisan, 2019-10-15:
That's really odd. Are you really sure you're actually writing the os to the sd card and not running some old image/setup?

jeanfbrito, 2019-10-16:
Yes. I have done this alot of times. I will try some older releases. The
last prerelease gives the same error.

ccrisan, 2019-10-16:
Can we see some screenshots/photos/logs?

glynneh, 2019-10-16:
Exactly the same issue here.
To start the RP3 is connected to the network hub, and plugged into a monitor (HDMI).
Starting the RP3 for the first time after flashing, every looks good and the Motioneye is reachable using chrome browser (USB camera is visible).
I then turned on the wifi option within gui, and entered my wifi details.
I the APPLY the changes and the pi reboots.
The wifi fails to connect, and loop with the same mentioned critical error.
I´ve tried reconnecting the network cable, but same error.
The SD has to be reflashed to fix (without setting up the wifi).

Screenshot attached.

I recently flashed a Pi Zero with a 
[image]

ccrisan, 2019-10-16:
@glynneh in your case it's `/data/etc/motioneye.conf`, instead of `/data/etc/motion.conf`. Either way, the issue is very strange. I'll try to reproduce the issue myself.

ccrisan, 2019-10-16:
Guys have you added any particular configuration file on the `/boot` partition?
@glynneh your system should not continue booting at all after a failed network. Have you configured your system a networkless?

jeanfbrito, 2019-10-16:
Nothing. Just flashed the SDCARD with the image. Booted and accessed by
wired network, configured wifi and reboot.
Then shows that message and keeps rebooting.
I use RufusUSB to flash the image to the sdcard.

glynneh, 2019-10-16:
Hi @ccrisan 
I made no changes to the downloaded image (latest downloaded today for the Pi3).
This only happens when I apply the wifi settings in the webpage GUI first time, then after a reboot, the looping of the error and failure of the wifi connection. Same a described by @jeanfbrito

ccrisan, 2019-10-17:
@glynneh something special about your WiFi configuration?

glynneh, 2019-10-17:
@ccrisan Not as far as I know. I have around 20 other devices that connect ok. Including MotionEyeOS running on a RPZero. I will try another AP.

ccrisan, 2019-10-17:
Could you maybe recover the configuration files from a faulty setup and attach them here (removing sensitive data)?

jeanfbrito, 2019-10-17:
In your Brand New setups in raspberry 3 It doesnt happens @ccrisan ?

jeanfbrito, 2019-10-18:
Installed version 20190427 and worked well. Then, updated to 20190911 and
is working fine.
I think that the problem is a brand new installation with the image of
version 20190911.

ccrisan, 2019-10-21:
> In your Brand New setups in raspberry 3 It doesnt happens @ccrisan ?

Nope, I couldn't reproduce this issue.

@userdsp does your error message mention `/data/etc/motion.conf` or `/data/etc/motioneye.conf`?

Guys, can you please recover and attach your `/data/etc/motion.conf` and `/data/etc/motioneye.conf` files from your problematic systems?

reolus, 2019-10-28:
I got the same error after setting wifi after clean image.  Raspberry pi new image. in the motioned.conf file there look to be 2 motion_binary entries without the file location following it - should be /usr/bin/motion maybe?

I'll try pulling a copy of the files again later.  Might try modifying as well

pipaul, 2019-11-01:
I can reproduce this error with a new install of motioneyeos-raspberrypi2-20190911.img.xz like this:-
1. Browse to the motionEye IP address.
2. Open Network and change to Static IP settings, don't do anything else.
3. Click Apply and reboot to get the error.
If the General Settings are applied first, the reboot is OK and the Network settings can then be edited OK.

jeanfbrito, 2019-11-06:
The solution as I said is to install an old version and then update to the
new one. Last stable release before this last one is ok.

ccrisan, 2019-11-08:
Could it be again a problem of browser's cache? Can you check that the Motion Binary option is set to `/usr/bin/motion` all the time?

mission-webstyle, 2019-11-13:
So i got the same problem and i checked the motion binary option and its set to /usr/bin/motion. Hope it helps =)

Hardware: RB 4 4GB
OS-Image: motioneyeos-raspberrypi4-20190911.img

ccrisan, 2019-11-13:
@mission-webstyle so what you're saying is that you get the `failed to read settings from "/data/etc/motioneye.conf": invalid configuration line: motion_binary` error but you have `motion_binary` set to `/usr/bin/motion` in your `/data/etc/motioneye.conf`? If yes, can you **attach** your `/data/etc/motioneye.conf` here?

kbarbora, 2019-11-15:
same thing for me...
any solution so far?
I tried with different release in a Pi3B+

jeanfbrito, 2019-11-15:
Invalidate the issue is the solution...

Install one release before, and then update to the last. It works.

Revilum, 2019-11-20:
I have the same Issue and motion_binary isn't linked to anything. 
[motioneye.conf.zip](https://github.com/ccrisan/motioneyeos/files/3869835/motioneye.conf.zip)

Revilum, 2019-11-20:
I installed an older version and a ton of new options showed up

pacomx, 2019-11-21:
Same issue here. Fresh install from latest "motioneyeos-raspberrypi2-20190911.img" using linux script: `./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos.img"`
It booted fine when connected through ethernet cable. Next, I configured WiFi using GUI, applied changes and this error appeared:
[image]

I fixed it by doing again a fresh install but preconfiguring a wireless network connection:
` ./writeimage.sh -d /dev/mmcblk0 -i "/path/to/motioneyeos.img" -n 'yournet:yourkey'`

TimBloom, 2019-11-22:
Same issue, same process to reproduced. Downloaded latest image, wrote image to SD, booted and worked.  Set up wifi and a static IP through the in-browser menu. Rebooted to an identical failure. Sounds like the WiFi or network setup piece is breaking this config file.

jasaw, 2019-11-24:
Can you guys please extract from your broken pi and attach your /data/etc/wpa_supplicant.conf here.

dborden0, 2019-11-24:
Exact same problem here. How do we extrace the /data/etc/wpa_supplicant.conf file from a windows PC?

virtualjp, 2019-11-28:
I've encountered this issue myself and been performing a bit of troubleshooting.

Seems that when you log in via the web frontend for the first time and enter the wifi settings it clears some of the Expert Settings (or does not propogate them to begin with) and you're left with an empty value in Motion Binary, which after an Apply and reboot results in the critical boot failure.

ccrisan, 2019-12-01:
Guys could you please try reproducing the issue with the *Expert Settings* section opened and see if the *Motion Binary* entry gets changed (by Chrome's autofill, for example) while you edit your options?

virtualjp, 2019-12-03:
Hi Calin, so I have just now re-flashed my card with a stock/default image, booted up using a wired connection and as soon as I have gone into the web frontend and expanded the Expert Settings, all the fields from Date HTTP Host to Motion Binary are empty and with the red exclamation mark.
Also, I am using Firefox as my browser so don't believe this to be an autofill issue.

ccrisan, 2019-12-03:
@virtualjp can you please make a screencast or somethink so we can see what's going on, in action? Also, could you open developer tools (F12) before stuff happens and see if you can capture some JavaScript errors or anything like that?

virtualjp, 2019-12-03:
@ccrisan Looks like this is more of a human error on my part, and may also be what happened for the OP and subsequent posters.
I was not entering in the admin creds when logging in to the GUI, but rather just clicking Login at the prompt, this then gives access to some of the settings but without escalated privilages and thus is not pulling in all the settings. One can set the WiFi details and then click apply which does then prompt for the admin credentials to complete the reboot, which then writes back the blank settings resulting in the behaviour encountered in this thread.

Modro1991, 2019-12-12:
I have also the same issue
Entered the GUI interface with ip adress.
entered my wifi ssid and password applied and rebooted.

ccrisan, 2019-12-15:
@virtualjp your hint helped me identifying the bug and it was indeed related to the fact that non-admin users are exposed to expert settings. This is no security issue though, for those who are wondering, since it asks for admin credentials before applying the settings.

Fixed via https://github.com/ccrisan/motioneye/commit/2caacf7f899070fc169ecdad5bc066214c1d1712.

closed 2019-12-15 by ccrisan

mauricioduarte01, 2020-01-26:
Hello, how Can i download the latest image for my Pi3 with the patch applied? Or I have to build from the source code? Thanks

ccrisan, 2020-01-26:
You can wait for the next pre-release, hopefully out by tomorrow, if everything works well.

GileraGFR, 2020-02-09:
Without patching or updating just make sure to login before making any changes.

wanradt, 2020-04-16:
@ccrisan You don't mention, how to recover from this situation. Do I need delete some files or replace something? It is unclear what happens and how to make all fine again.

starbasessd, 2020-04-16:
You can grab and reimage with 20200203, and that will resolve the issue.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

wanradt, 2020-04-17:
@starbasessd actually I'd prefer to make corrections where needed, but I don't see such images as you suggest on [list of supported devices page](https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices), every device has listed 20190911

starbasessd, 2020-04-17:
https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

snowflakeas, 2020-07-26:
same issue - all the links on https://github.com/ccrisan/motioneyeos/wiki/Supported-Devices are old(!)

starbasessd, 2020-07-26:
I use this link:
https://github.com/ccrisan/motioneyeos/releases

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

snowflakeas, 2020-07-26:
thanks, would be even better if support-devices page could be updated (because of google)

starbasessd, 2020-07-26:
@ccrissan needs to update, or get someone to update a lot of the pages, not
just that one.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomhorak, 2020-09-05:
I still get this issue with 20200203. D/L image, flash SD, create wpa_supplicant.conf and ssh files on card.  Boot pi zero W and it panics into reboot loop.  Then just tried the image with no other changes.  Same thing.  I think the image write is perhaps not working entirely.  I'm going to format my card before image this time.

starbasessd, 2020-09-05:
Be sure your PiZero is a PiZeroW or PiZeroWH, the plain version doesn't
have wifi.
Double check your wpa_supplicant.conf file.
Do Not create it with Windows Notepad.
Use a Linux friendly text editor like NotePad++.
Here is a sample file, change Country, ssid, and password to yours. All
three are case-sensitive. Quote marks are required.
-----
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
    ssid="your_ssid"
    psk="your_password"
key_mgmt=WPA-PSK
}
-----
Let me know if these help.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

tomhorak, 2020-09-05:
Yes, I have 2 Zero W's and happens with both.  This time, I just did a plain vanilla image, no files let it boot.  Then logged in as admin, changed password turned on wifi, and set that up.  Rebooted, everything works just fine.  I could see the encoding being an issue, but I used UltraEdit32 and Unix file.  Its something to do with adding a file/files, odd.  If I had more time I'd help troubleshoot, but I have to run and this is working now!  Makes no difference which version btw.