motioneye-project/motioneyeos #2148: Pi Camera V2 Not showing image

Philster72, 2019-10-16:
Hoping someone can help

Have just done a fresh install of motionEyeOS 20190427 on a Raspberry Pi 3B+.  I have the Pi Camera V2.  The camera was previously working on Raspbian os with MotionEye running, but decided to try MotionEyeOS.

Basically after logging into the Web UI, the camera view just has the spinning wheel and I never see an image.

Not sure what I am missing.
Log files attached.

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3732883/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3732884/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3732885/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3732886/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3732887/motion.log)

Any help would be greatly appreciated.

Regards
Phil

jasaw, 2019-10-17:
Are you using Chrome web browser? Try the latest MotionEyeOS release: 20190911
https://github.com/ccrisan/motioneyeos/releases

Philster72, 2019-10-19:
I have updated to the latest release and all appears to be working now.  Thanks so much for your help.

closed 2019-10-19 by Philster72