motioneye-project/motioneyeos #2151: Camera keeps freezing

ir-fuel, 2019-10-17:
I have 3 Raspberry Pi's at home running MotioneyeOS. They all used to work perfectly, but all of the sudden on of them has the camera that freezes. The rest of the system is responsive (web frontend, ssh, etc).

Below the necessary info:

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL. 

My camera model is: Raspberry Pi camera.

### Network Connection

My motionEyeOS unit is connected to the network via:Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

### Log Files

This is the relevation portion of `motioneye.log`

```
2019-10-17 10:27:05: [motioneye]     INFO: rebooting
2019-10-17 10:27:05: [motioneye]     INFO: executing "/sbin/reboot"
2019-10-17 11:18:16: [motioneye]     INFO: hello! this is motionEye server 0.41
2019-10-17 11:18:20: [motioneye]     INFO: cleanup started
2019-10-17 11:18:20: [motioneye]     INFO: wsswitch started
2019-10-17 11:18:20: [motioneye]     INFO: tasks started
2019-10-17 11:18:20: [motioneye]     INFO: mjpg client garbage collector started
2019-10-17 11:18:20: [motioneye]     INFO: smb mounts started
2019-10-17 11:18:21: [motioneye]     INFO: server started
2019-10-17 11:28:41: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-10-17 11:28:51: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2019-10-17 11:29:01: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
```

The last couple of lines just keep repeating.

Any idea what might be happening?

Thanks!

gb-dursley, 2019-10-21:
I suspect a hardware failure in the camera. Have you tried swapping the camera in question with either of the other Pis. If the fault doesn't move with the camera, you've proved it's not the camera. If it does, it is.  Electronics are subject to 'infant mortality' - the rate of failure decreases exponentially over time; so most failures occur in manufacture, either of the dead chip or the end-product. Most failures for end-users occur within the first few seconds or minutes.
Have you tried a fresh install of the OS (after a backup of the settings for that Pi)?

ir-fuel, 2019-10-21:
This camera has been running for months without issues.
I first tried a normal update to the latest version, but that didn't change anything.

I did an update to the latest prerelease yesterday. Seems to have solved it for now, but I'll monitor it for a couple of days before stating that it really works.

ir-fuel, 2019-10-22:
And it's hanging again since this afternoon :(

ir-fuel, 2019-10-22:
What's strange is that the camera seems to keep on recording. I have it in permanent recording mode and I see on the storage that videos are being added all the time, even right now while the live image is frozen.

gb-dursley, 2019-10-23:
Ben there, experience that.  Too much 'noise' on WiFi.  
Our house has massive stone walls so WiFi is a bit hit and miss anyway.  However, until a few days ago, I had an experimental camera sited so that both the cordless phone base-station and the wireless thermostat/remote control for the central heating were between the laptop, camera (~ 8 feet apart) and the router; the frame rate on the live feed is lower (and varies considerably) and when the controller kicks in (or next door switches on his router), there's a complete loss of picture for several minutes.  Solution was to re-arrange the Wifi devices, so that the camera and laptop are not between C/H thermostat and boiler controller and well-away from the cordless phone base-station.  Freezing problem gone and although the live-feed frame-rate has improved it still stutters (frame-rate and video).

ir-fuel, 2019-10-23:
That can't be it. I have a 2nd Pi + camera sitting right next to this one, and that one works flawlessly

gb-dursley, 2019-11-01:
So what are the differences between the two? What has changed with the one that freezes? Given that the 'faulty' Pi is still recording Video OK, and the motioneye.log message, it has to be something to do with streaming video from this Pi over the the WiFi connection. 
You say "the 2nd Pi [is] sitting right next to" the Pi with the intermittent fault; could it be significant that it isn't in _exactly_ the same place? Some occasional source of low-level interference very close to this Pi?  Mains cable/lead to something noisy (e.g. CFL or LED bulb in table light)?  
Many years ago, when networks were rudely known as notworks, a client of mine experienced a total network (co-ax Ethernet) crash most mornings at the roughly same time - when the next-door factory used an ancient arc-welder for an hour or two, causing HUGE EMI spikes on every cable within a radius of several hundreds of yards.

Solderfinger, 2020-07-06:
Sorry for reviving this rather old thread.
I have the exact same problem with four rpi zero w cameras of which one freezes regularly. The system has an additional master rpi (rpi 3B+) that collects the wifi streams and displays them. With the frozen camera picture i can still logon to every rpi or press keys or click on the motioneye logo on the master rpi to restart. Also, while the picture is frozen on one display, i can logon from another computer and see four non-frozen pictures.
While i do not exactly understand what is going on i think about soldering on a connector for an external antenna. Hopefully it will then work flawlessly...

jasaw, 2020-07-09:
@Solderfinger I highly suspect it's wifi related. The rpi zero W only has 2.4GHz wifi. 2.4GHz wifi struggles when you push multiple high resolution high framerate video streams over it. Try running the four rpi zero W in RTSP mode with 1 Mbps bitrate. Try lowering the resolution and framerate to try isolate the problem.

gb-dursley, 2020-07-09:
I've had this problem too (with both a single camera and 2) and came to the same conclusion as **jaswaw** suggests, a network traffic issue.  Having done lots of experimenting, I bought another a router (Edimax) dedicating it and its single network to the job, siting the router where it has a clear line-of-sight to both PiCams. Over several days I then continuously monitored the networks 'visible' to the router (with LinSSID) to pick a trio of adjacent channels not currently used by a detectable network -  LinSSID GUI helpfully shows the overlap from networks on adjacent channels.  The 'quietest' location is on a window-sill where the thickness of the walls minimises the signals from routers neighbouring properties whilst maximising the signal from the two PiCams!
Then I started systematic experiments with all the parameters which might reduce the network traffic.  It's essential to be systematic, recording the changes (one at a time only!) made and the resulting effects over a period of time.  Tedious, but essential.  IIRC (I can't check easily), cutting the frame rate to 30 fps and the resolution to ½ the cameras maximum did the trick.
This set-up has been rock-solid for some 8 months without _any_ tinkering bar adjusting the mask.  The recordings are perfectly acceptable and a lot better than many off-the-shelf CCTV systems.  During my experiments, it did seem that manually deleting unwanted footage might be causing freezes - the solution being to set-up motionEyeOS to delete all clips older than (IIRC) 1 month - long enough to keep anything that might be wanted without risk of filling the available space.  
BTW, it's asking for trouble to use anything but a high-speed (C10 U1) High Endurance 32 GB microSD Card of a reputable make.  I've got 4 Kingstons, 2 for the PiCams, 2 for my dashcam (1 in use, 1 spare in the glove-box).  The dashcam hammers the card with continuous loop-recording and when the car is parked in the sun, the cards get roasted!  All 4 cards still working fine, the oldest (dashcam) bought 1st March 2019.

cjh7583, 2021-05-11:
I know this is an old thread but had exact same problem where cameras where randomly freezing throughout the day.
I added a USB hub and ran the cameras from that and seems to take the load of the CPU on the Pi and no more freezing.
Hub is just an amazon basics one and works fine.