motioneye-project/motioneyeos #2155: [Feature Request] Prebuilt image for embedded x86 systems (#3)

Giga-Pudding, 2019-10-20:
Hi ccrisan,

I know [you have answered](https://github.com/ccrisan/motioneye/issues/619) this question already two years ago, but I would like to give two good reasons for providing a generic x86 image of Motion Eye.

The first thing is, that the currently supported hardware is (mostly) not powerfull enough, for the flawless processing of high resolution / high FPS Video. At least that was my experience. The second point is, that providing an x86 image would allow people to recycle old Hardware like Thin Clients and stuff.

I really prefer small single-board devices over fat old computers, but i think there are some good reasons though...

What do you think?

BR,
Giga

bobafetthotmail, 2020-06-03:
it's probably easier to add a package for it in OpenWrt packages repo https://github.com/openwrt/packages

ZeroDot1, 2020-09-10:
Hi @ccrisan, I have enough old hardware that I could use directly with Motioneye, it would be especially good if you could just install it in ArchLinux and use it directly.

bobafetthotmail, 2020-09-10:
> Hi @ccrisan, I have enough old hardware that I could use directly with Motioneye, it would be especially good if you could just install it in ArchLinux and use it directly.

motioneye is in the AUR so you can install it from there if you are on Arch Linux

ZeroDot1, 2020-09-14:
Hi @ccrisan, I asked some friends, they would use Motioneye immediately if there was a standard x86 image, because then they could use it directly with their old notebooks.