motioneye-project/motioneyeos #2161: Motion notification failed

joeydream, 2019-10-22:
Hi, everybody

I failed in setup for motion notification and it was told:

Notification email failed: 5.7.8 Username and Password not accepted. Learn more at 5.7.8 https://support.google.com/mail/?p=BadCredentials q25sm3329396wra.3 - gsmtp!

here is my setting, can anyone help? thz
[image]

Dewey3, 2019-10-23:
You should drop the "@gmail.com" from the SMTP Account name, Gmail doesn't use that for the account name.  Also, I don't know if it will cause the problem you're having, but your from address should have a valid email address.

rolfep, 2019-10-25:
It might be a matter of enabling "less secure apps".  https://support.google.com/mail/thread/5621336?hl=en

My gmail address is the SMTP account value, as the mouse-over help suggests.  The from address can be left blank/is that same gmail address.  In my experience.  Also, one arcane switch might need to be flipped by going to https://accounts.google.com/DisplayUnlockCaptcha and "To allow access, click the Continue button below".  That's done the trick for me a time or two.  Thank you, Godgle!