motioneye-project/motioneyeos #2166: Lost what settings to set to get maximum performance/fps from Pi Zero 0

invisible999, 2019-10-24:
Ladies and gentlemen,

I do apologize in advance for asking this question but could someone point me what needs to be done and in which order to get maximum fps from PI Zero W?

I've downloaded and  installed the latest build. After setting up PI Zero W and local v1 camera as MIMAL camera, on the preview/right camera window I can't get more than 0.5 fps and the same is for recording.

There are dozens of 'pi zero low fps' topics all around the internet. But I'm struggling to understand which one is relevant. Basically I want two things:

- Do motion detection and record videos the the highest fps possible with resolutions either 800x600 or 1024x768

- Have highest fps video streamed when logged on to the web interface.

Is this possible to archive?

Thanks

jasaw, 2019-10-24:
To encode video at around 15 fps, you will need to select the OMX variant movie format.

Regarding live stream via web browser, a pi zero doesn't have the cpu to do high fps.

jasaw, 2019-10-24:
>    Is this possible to disable live streaming with Motioneye/motioneyeos while having only motion activated videos recorded by the system or motion detection is the result or analysing the live stream?

There's no way to disable the live stream. However, the live stream is only working when you have a web browser connected to the web front end. In simple words, just close the web browser and live stream won't take up any CPU resource. Motion detection and video recording will keep working even if web browser is not connected.

>    If I only watch live stream and disable all motion detection, will it have beneficial effect on fps from web login?

Disabling motion detection will only gain you a little bit of frame rate.

>    on web interface, after I did re-imaging the SD card now I see 4-4.5/8 fps as stream/capture. Is this normal for PI Zero 0?

5 fps is normal on a Pi Zero for 800x600 resolution. The frame rate is highly dependent on video resolution and wifi signal strength.

>    If I do not use locally attached camera on Motion/Motioneyeos and use it only as a server for remote motioneye or rtsp streams, will I get a higher frame rate? What can be done to optimize performance if the PI is used only as a head server without locally attached cameras?

H264 over RTSP video stream requires a lot of processing power to decode. You won't be able to get satisfying performance out of Pi Zero receiving RTSP.
If you want to receive multiple RTSP streams, you need a minimum of Raspberry Pi 3B+, or RPI-4B, or Odroid XU4. _Disclaimer: I don't have such a set up, so this is just my gut feel._ Some IP cameras have dual RTSP streams, one high resolution, one low resolution. You could use the low resolution for motion detection, and record the high resolution (pass-through recording feature, but I'm not sure MotionEye currently supports it or not).

invisible999, 2019-10-24:
Great, thank you very much for insights, would you mind to ask couple of more clarifying questions?

H264/RTSP - is this memory or CPU intensive operation for following cases?
 
a) remote PI with camera which sends the stream to the the head server
b) the head server which does not have a camera but acts as a stream aggregator and motion detector

You mentioned that a lot of processing power is required to decode stream. Is it hardware/gpu accelerated when motioneyeos is used? 

I am asking because it is only CPU intensive, then I should be getting PI 3 A+ model. As I understand it has the same 4-core CPU as B+ but lacks number of USB ports  which won't be needed in my case.

jasaw, 2019-10-25:
For remote pi sending rtsp stream, if you run the latest prerelease and run fast network camera mode with rtsp protocol, it doesn't use much of the cpu at all. All the heavy lifting is done in the GPU.

For the steam receiver, decoding is done in cpu even on motioneyeos. I don't know how many streams a pi can handle.

invisible999, 2019-10-25:
Thanks for clarifying... Why decoding is done by CPU and not GPU, any reason for it?

If I want to setup following system - central PI without camera receiving streams from 2-3 remote PIs and doing detection, what would be the optimal settings? Should remote cameras add as motioneye remote cameras or they should setup as network cameras?

Separate, but related question - is motioneye/motioneyeos as fast for stream sending as alternative/native methods? I came up to the following link
https://chriscarey.com/blog/2017/04/30/achieving-high-frame-rate-with-a-raspberry-pi-camera-system/
where the author claims 

> Using this technique below, my camera system is now able to get closer to the full resolution for each Raspberry Pi camera which is 1920×1080 at 30 frames per second – and reliably running for years without issue.
> 

Can something like this done with motioneye?

jasaw, 2019-10-26:
There are a few reasons why decoding is done in CPU.
1. The benefits of decoding in GPU is not clear. If decoding then encoding in GPU, we are going to hit memory bandwidth bottleneck, i.e. both encode and decode will suffer from low frame rate. The Raspberry Pi 4B has much higher memory bandwidth, but I haven't looked into it yet.
2. `ffmpeg` lacks any sensible API to select which video decoder to use in motion software.

Regarding your question on the fast stream on Raspberry Pi, it can certainly be done in motionEyeOS. This is the Fast Network Camera mode (with RTSP protocol) in motionEyeOS. The latest pre-release is even better because it outputs high resolution RTSP for recording (pass-through recording feature in motion), and low resolution MJPEG for motion detection, all done in GPU. In the blog link you posted, the author was using a much more powerful machine to receive the RTSP streams. You could do the same thing with MotionEyeOS (Fast Network Camera) + Powerful machine as RTSP receiver.

invisible999, 2019-10-26:
Great, seems thanks are developing in the right direction. I just need to wait until release to be made official. 
When is ETA, do you know?

pdisner, 2019-10-26:
So I take it that you already know and have tried to adjust thru the Motion software? I, a new user also. I cant see, to get all my cams to show all on one screen. I’m aware of the “add camera” option at the top left corner of screen. Just have to keep trying.  I also have no idea on how to get the camera to get past my router so that I can remote access.  Any luck with Hess?

Paul Disner
(520) 307-6844
pdisner@me.com

jasaw, 2019-10-26:
I doubt there's going to be a stable release anytime soon because we have just added some new features recently. I recommend you try the pre-release.

@pdisner please don't hijack this thread. Please open a new issue.

invisible999, 2019-10-27:
> I can't see, to get all my cams to show all on one screen. I’m aware of the “add camera” option at the top left corner of screen. Just have to keep trying. I also have no idea on how to get the camera to get past my router so that I can remote access. Any luck with Hess? 

Could you tell:
- the model of hardware you are using
- the motioneyeos software version
- ifre local hardware and the remote hardware in the same network?
- how you tried to add camera - which option did you choose?

>  I also have no idea on how to get the camera to get past my router so that I can remote access

Above above question - are you asking to have remote access to the motioneyeos web interface or to add remote camera to motioneyeos when camera is behind router?

For former, when hardware is behind the router, it entirely depends on router itself. You need to  check router manual and see how to properly setup port redirection which will map router's WAN interface IP address and port to motioneyeos IP address and port (80 by default). 

After setting up port redirection you can access to motioneyeos by using http://your_routers_wan_ip:the_port_you_setup_for_port_redirection

invisible999, 2019-11-03:
OK, one more update. 
I spent considerable amount of time reading about encoding, decoding, GPU usage and so on.

I do love MotionEyeOS for simplicity and functionality. However, seems even with PI 4's performance won't be enough to do processing of two RTSP 720p streams from remote IP cameras. As it was mentioned in other places, memory bus most likely will be the choking point.

So I pulled the trigger and ordered Nvidia Jetson Nano development kit. According to specs, it has ability do decoding 4K stream at 60fps which should be more than enough to process two 720P stream but do object/face detection/classification.

While I plan to use MotionEyeOs for two PI Zeros sending stream to Jetson Nano, the nano itself should run different software to fully utilize its potential.

BTW, if anyone here have used Nano for such purpose, feel free to drop me a message and share your observations/experience.

Thanks again.

popoviciri, 2019-11-03:
Hi @invisible999 
I see @jasaw explained you how to take advantage of the latest rtsp/mjpeg streaming capabilities which are part of the last builds and I though of detailing my setup, for reference. Obviously, won't do face detection and high fps in the frontend.
- setup the rpizeros RTSP with low res / low fps mjpeg streaming;
- on the machine running motion, the cameras can be added by using the MJPEG stream instead of RTSP: `mjpeg://IP-addr:8081/`; in this case the low res/fps stream will be used for motion detection, using minimum of resources.
- adding `netcam_highres rtsp://IP-addr/h264` and `movie_passthrough on` in the extra motion options box, it doesn't do any RTSP processing but simply dumps the RTSP stream on the disk, when motion is detected.
I have 4 rpizw + 2 rpi1 streaming RTSP at 1640x1232and 15fps and 2Mbps, together with MJPEG at 320x240, 2fps and 0.8Mbps. The two rpi1 are idling at roughly 80% and rpizero at 70% CPU. All the mjpeg streams are captured on a rpi3 B+ running motioneyeos with the mentioned options. This last box uses less than 20% CPU while having motion enabled for all cameras and recording simultaneous RTSP streams. Uses some 600MB of memory though.
I also went for the power hungry machine (8 cores I7) for motion detection and RTSP decoding, but I ended up concerned about the CPU temp and found that's really not needed.
Damn, that Jetson Nano looks tempting! For other projects though.
Hope I'm not off-topic! Cheers!

invisible999, 2019-11-03:
Hey, thank you very much for write up and instructions. Once I get my stiff from Arrow and Aliexpress, I will try to set them up and then will report here as well. I will also try to setup MotionEye (not MotionEyeOs) on Jetson to see how/if it can perform for such role.

So once I get my hardware, here is how the central server and remote PI Zeros should be setup:
- Code for PI Zeros: need to run latest pre-release build
- Setup PI zeros as Fast Network Camera - no processing is done, they just send streams
- Setup central server with mjpeg://IP-address:8081 for motion detection
- Setup central server - in extra motion options settings add netcam_highres rtsp://IP-addr/h264 and movie_passthrough on

Does the central server also need to run latest pre-release code or current release should work as well? Also, the central server should run motioneyeos or Raspbian + Motion

popoviciri, 2019-11-03:
That looks right! Here is the reference to `netcam_highres` option in motion: https://motion-project.github.io/motion_config.html#netcam_highres.
The main hub doesn't need to run the latest pre-release, in fact any distro running motioneye + motion would work. I prefer motioneyeos on the raspberry pi 3 as main server, due to the low power consumption and it's read-only robustness. I can unplug and plug my cameras back in as many times as I want. They always start right up! Cheers!!

Sakthivel-bpigt, 2019-11-06:
Hi @jasaw 
you replyed to @invisible999 **first** question as follows
> There's no way to disable the live stream. However, the live stream is only working when you have a web browser connected to the web front end. In simple words, just close the web browser and live [...]

I have one more follow up question regarding this;
does the "disabling the live stream" help improve the recorded videos quality in terms of FPS as well as resolution?
I'm planing to use this recorded video to extract information using machine learning, just wanna know(just curious) before i step into it

jasaw, 2019-11-06:
Disabling live stream generally gives you slightly higher framerate. You'll have to test it yourself whether the recorded videos are suitable for your machine learning purpose.