motioneye-project/motioneyeos #2167: motionEyeOS 20190911 Color/Whitebalance Issue

davidaames, 2019-10-24:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: MMAL

My camera model is: One of those Chinese 5MP Nightvision cameras with 2 IR lights on the side.

After updating to 20190911 my nightvision camera, which is placed outside displays everything in a red hue. I observed it struggling to color balance today for about a minute, twitching from regular colors to the red hue, and staying on it.

popoviciri, 2019-10-27:
If you're using motion detection (default install), you must add `mmalcam_control_params -awb greyworld` to "Extra Motion Options". If you're using the "Fast Network Camera" (FNC) mode, then the `Greyworld` option is available in "Automatic White Balance".

AndyMAu, 2019-10-30:
Thanks popoviciri, I was wondering why everything was red. I have applied the command and all good now.

jasaw, 2019-10-30:
@AndyMAu Are you happy to close this issue?

davidaames, 2019-10-30:
Thanks for your input. Would be awesome to include options like this in the GUI for future releases.

jasaw, 2019-10-30:
@ccrisan What do you think of exposing mmal controls on the web interface, similar to FNC type controls?

ccrisan, 2019-11-01:
@jasaw I'm not sure what that implies. Is there a command that lists them and allows tweaking them?

jasaw, 2019-11-01:
It's just exposing whatever options that  mmalcam_control_params supports, but I think it's more trouble than it's worth. Maybe the best solution is to document all  mmalcam_control_params options in the wiki.

AndyMAu, 2019-11-07:
> @AndyMAu Are you happy to close this issue?

I was merely commenting about the issue. My only other question is why has this never been a problem before for me with previous releases?

Dr-Ari-Gami, 2020-05-29:
Where can I access "Extra Motion Options"? I'm running motioneyeos on a Rasberry pi 0 W. For the life of me I cannot find it.

I have the pink tint issue as well.

Solarbotics, 2023-08-27:
I'm sure this he's figured it out, but for future answer-seekers, it's under the "Video Device" section, below the "Frame Rate" slider