motioneye-project/motioneyeos #2169: Pi ZW camera module not detected

SichongP, 2019-10-27:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI ZW

### Camera

I am using the following type of camera: (Supposedly MMAL as it was claimed to be pi camera module)

My camera model is: https://www.amazon.com/gp/product/B074DY544H/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
N/A

### Log Files

[messages.log](https://github.com/ccrisan/motioneyeos/files/3775544/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3775545/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3775546/dmesg.log)

Hopefully this isn't a duplicate. I have browsed through several similar issues here but couldn't find a solution.

From the webpage interface, the camera can only be detected as v4l2 camera but not mmal. Although when I chose v4l2 it said failed to open device (log files attached)

From terminal, I don't have /dev/video0 (I'm using motioneyeOS):
```
# ls /dev/video*
/dev/video10 
/dev/video11 
/dev/video12
```

And:
```
# v4l2-ctl --list-devices
bcm2835-codec (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

Cannot open device /dev/video0, exiting.
```
When I run `vcgencmd`:
```
# vcgencmd get_camera
supported=1 detected=0
```

I have checked ribbon connection and power supply. Camera has red LED on.

Any idea what could be the problem?

SichongP, 2019-10-27:
Update: I just tried on raspbian and had exact same problems. Looks like this isn't related to MotionEyeOS. I'm leaving it open for now in case anyone had similar experience. Otherwise please feel free to close.

gerald-romero, 2019-10-30:
I have had issues with different pi cameras where the very small cable from the camera to module is not seated properly. Not the cable from module to pi. Bear in mind that even after re-seating the cable and getting the camera to be detected, it was almost always intermittent. Some just seem to be poorly assembled or something.

robevansuk, 2019-11-15:
Had a similar problem spinning my set up today. Had the ribbon cable in the wrong way around. Double check and its good now.