motioneye-project/motioneyeos #2173: Monitor station with rpi and touch screen

SerialIterator, 2019-10-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

jasaw, 2019-10-30:
@SerialIterator I'm not exactly sure what kind of set up you are after, but I'll try my best to answer your question.

To me, a typical security system looks like this:
```
     multiple IP cameras     ----->        headless server          ----->          optional display
(MJPEG or RTSP video streams)      (motionEyeOS or motion+motionEye)      (raspbian or any device with a screen)
```

The IP camera can be an off-the-shelf IP camera (as long as it's supported in motion), or it can be motionEyeOS running Fast Network Camera (FNC) mode. Keep in mind that if you run FNC mode with RTSP, you won't get a camera preview on the HDMI port.

As for the headless server, you should run motionEyeOS on a more powerful device because it has to receive multiple video streams and do motion detection etc. The alternative is to run motion & motionEye on any Linux OS on any hardware you like, PC or old laptop, or Odroid, or Raspberry Pi.

The display that shows all the video feeds on the same screen is optional. If you choose to have a display, you have a few options:
1. Use a tablet and point the web browser to your headless motionEye server. You'll have to configure this tablet to leave the display on, and always plugged into power source. I'm not sure what the performance is like. Alternatively, you could look into motionEye smartphone app. I have not tried the app myself, so can't tell you much more.
2. Combine the display functionality into the motionEye server itself, like what I described in my other post. You attach a screen to your server, auto-start the web browser, and point it to your motionEye server, which is localhost in this case.
3. Set up another device with a screen (could be raspbian + screen, or old laptop, or anything you have available). Point the web browser to your headless motionEye server.

SerialIterator, 2019-10-30:
Thank you for giving an overview of the different configurations. My intended setup is:

rpi zero w's as remote cameras running motionEyeOS in FNC mode. 
rpi 4 as a server connected to a 7" touch display.

Based on what you've stated, I would need to load the rpi 4 with raspbian and collect and serve the camera feeds using motion+motionEye. Then run a web browser at start up pointed at the localhost to display the feed to the connected touchscreen over hdmi.

Would motion+motionEye running on the rpi 4 enable me to use motion detection and save motion detected files to cloud storage? I also read that I would need the nightly build to use an rpi 4, or is that for motionEyeOS only?

jasaw, 2019-10-30:
motion+motionEye on Raspbian should allow you to do motion detection, record video on motion, upload to cloud. You shouldn't need a nightly build for RPi4.

closed 2019-11-02 by SerialIterator

starbasessd, 2021-01-06:
> I have a relevant question, i have rpi4 with motion eye os, the pi has
> touch screen case, so its rpi case that comes with touch screen within
> itself, my goal to use the touch screen to show me my cameras feed, is
> there a way to have it working like that ?
>
> Thanks

If it is using a GPIO screen, you'd have to add the drivers for it
(waveshare drivers) which is not easy, and not included in the build.
If it is the hdmi screen, it can / will display, but does not have the
touchscreen drivers in the default build.
I would recommend running the motionEye install in RaspberryPiOS where you
can add the needed driver(s).

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ramybt, 2021-01-06:
Its gpio touch screen
So if i install the driver you mentioned, will that help to display my cameras feed/ motion eye UI on it ?

starbasessd, 2021-01-06:
Since the motionEyeOS is Read Only, you have to jump through a bunch of
hoops to install it, and you need to know how to make it read/write, move
the files to the proper locations, then make the file system read only
again, and there is no desktop to install, so the touchscreen is really a
moot point.
Your best option, unless you want to use a CSI camera, and hdmi output, is
to go with motion/motionEye on RaspberryPiOS.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ramybt, 2021-01-06:
👍🏼