motioneye-project/motioneyeos #2176: ERROR: invalid literal for int() with base 10: '5003.5'

MarcoMagana, 2019-10-30:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190427.
INFO: hello! this is motionEye server 0.41

### Board Model

I am using the following board/model: Raspberry PI 4 2GB.

### Camera

I am using the following type of camera: V4L2 .

My camera model is: USB Endoscopic camera.
[    1.625998] usb 1-1.4: New USB device found, idVendor=038f, idProduct=6001, bcdDevice= 5.38
[    1.626014] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.626027] usb 1-1.4: Product: USB 2.0 Camera
[    1.626039] usb 1-1.4: Manufacturer: lihappe8 Corp.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
N/A

### Log Files

I consider the following log files relevant to this issue:
motioneye.log

I have a Raspberry Pi 4 as a CCTV-Hub with 4 cameras connected. 1 USB camera and 3 MotionEyeOS remote cameras (Raspberry Pi Zero W with CSI PiCams V2.0)

Playing with the white balance slide control I wasn’t able to get back to the default value (5000), only close enough (5003). I decided to try with the arrow keys to get back to 5000, never lucky enough!, as the arrow key steps are to coarse. Then I held the Shift key down and pressed forward arrow key. MotionEye web page went blank and came back with an error message and prompted me to refresh the page. After refreshing the web page same error and refresh message appeared. None of the cameras was listed, nor the video previews. Checked the remote cameras visiting their own IP address and all the three were still working. Disconnected the USB camera and shutted down the Pi4. Reconnected everything and the same error message popped out. Did ssh to the Pi4 to see the camera is recognised there, all good so far. Checked several log files and found in motioneye.log the following message and backtrace.

ERROR: invalid literal for int() with base 10: ‘5003.5'
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1548, in _stack_context_handle_exception
  File "usr/lib/python2.7/site-packages/tornado/web.py", line 1756, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 270, in get
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 205, in wrapper
  File "/usr/lib/python2.7/site-packages/motioneye/handlers.py", line 317, in get_config
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1228, in motion_camera_dict_to_ui
  File "/usr/lib/python2.7/site-packages/motioneye/config.py", line 1228, in <dictcomp>
ValueError: invalid literal for int() with base 10: ‘5003.5’

So the Shift + → increased the value by 0.5 only. As the white balance value is expected to be an integer que config.py makes an exception error and stops working.

Can you add a value check or cast to the slide controls so these values are always Int?

Now I need to know where this values are kept so I can manually set it back to 5000 to recover my CCTV-Hub. Other wise will need to reformat the SD card to start all over again.

Thanks!

MarcoMagana

MarcoMagana, 2019-10-31:
I found it!

It’s in the camera-N.conf file. I edited to hold 5000 instead of 5003.5, rebooted and got back to the way it was.

MarcoMagana

MarcoMagana, 2019-10-31:
I found it!

It’s in the camera-N.conf file. I edited it to hold 5000 instead of 5003.5, rebooted and got back to the way it was.

MarcoMagana

adashrod, 2022-12-31:
I just encountered this same issue by using the arrow keys to change the `Rotate` value.