motioneye-project/motioneyeos #2181: How to lower the temperature on a Pi3B+

Eska72, 2019-11-02:
Hope that someone has an idea: I'm using MotioneyeOS for a long time now and having problems for a long time now. Three different systems with the same issue: Raspi turns itself off, lost connection etc. Reading the log files tells me that i lost the connection with the cams, or lost connection with the externel USB storage all that kind of things. First i thougt of power losses, instable adapter or something but since today i got a new message: throttlewatch: currently: arm-frequency-capped
Searching through all the issues i put on system monitoring on twp Pi's and one system tells me 79 degrees and the other says 84 degrees. Now i stopped recording the movies and lower the fps to 10 to both systems. In one of the two systems the temperature dropped to 69 degrees. The temperature icon in the top right of the hdmi screen i attached to the system has also disapeared.
The other systems needed to reboot but has totally lost its connection, i think because a high temperature, 83 degrees that it won't reboot through an interface and better works when its cooled down. (that system is not nearby for me so i have to manually reboot that one by pulling out the power cable).
Now my question: what is a normal temperature for a Motioneye system with 3 Foscam 720P camera's and continious recording to a externel usb storage? How can i make this a stable system?

Any idea's?

gr
Steve

vpmotion, 2019-11-02:
I personally have found that the motioneyeOS generates a lot more heat then the Motioneye installed alongside raspberian buster. Do you have any active cooling on the raspberry? The raspberry might be getting hot due to it continuously recording.

Eska72, 2019-11-02:
No i haven't an active cooling on the raspi. I diasbled the ftp, samba and the ssh server right now, set the memory to 512, csi camera led is off and there are no recordings for now. The temperature dropped to 65 degrees. Is there any other way to drop down the temperature besides an externel cooling fan?

vpmotion, 2019-11-02:
Do you have any cooling on the raspberry? Just a stand alone raspberry without any cooling does create a bit of heat. The best I can suggest is to put some cooling on the raspberry and make sure there is no restriction on air circulation.

Eska72, 2019-11-02:
No i got no cooling on the raspi, just standalone raspi with in a case. Do you gave any tips how to cool the raspi?

vpmotion, 2019-11-02:
https://coolcomponents.co.uk/products/aluminium-alloy-case-with-dual-cooling-fan-for-raspberry-pi-3-b?variant=14165207449661&currency=GBP&gclid=EAIaIQobChMIna3l7OPL5QIVFeDtCh0RMA_pEAQYAyABEgIcTvD_BwE

That is what I have and not a bad price. You can go for normal heat sinks as well just make sure they are pure copper or aluminium for proper heat dissipation and let it open and not in a confined enclosure.

Eska72, 2019-11-02:
thx, i'll try that hope it helps. í'll let you know if it solves my problems with the raspi and motioneye

jasaw, 2019-11-03:
Disable the HDMI port if you don't it. Underclock the GPU and cpu.

Eska72, 2019-11-17:
We're two weeks later and this is the result: i bought a new case for my Pi3b+ wich has a cooling fan integrated in the cover, cost about 8 Euro's. Temperature is lowered, it's 46 degrees now instead of 70-75 degrees! Using 1 camera, frame rate 15, recording  mp4 movies to an external disk with 75% movie quality. No problems for 5 days now. GPU Memory 512. All services on
So i think my problem is solved, Raspberries needing the airflow just as an desktop pc. And for sure if you use Motioneye. 

Thx everyone!

momciloM, 2020-01-25:
This is not a valid topic for this forum.