motioneye-project/motioneyeos #2182: Slow RTSP stream (only) in MotionEyeOS

rvdeijk, 2019-11-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera

My camera model is: Foscam FI9900P (V4)

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet (1gbit)

### Peripherals

I am using the following peripherals that I consider relevant to this issue: Netgear switch gs105e

For some reason the camera doesn't update that frequently and shows a 8/20 fps ratio. When I grab the RTSP stream url and paste it in for example VLC the stream is much faster. What am I doing wrong in MotionEye? I could provide logs if that's necessary, but which one is needed?

hexordecimal, 2019-11-15:
hey @rvdeijk i believe changing the configuration in /etc/motioneye/motion.conf to turn tcp off is the answer to your problem, 

```
webcontrol_interface 1
webcontrol_port XXXX
setup_mode off
webcontrol_parms 2
webcontrol_localhost on

netcam_use_tcp off
```

rvdeijk, 2019-11-16:
> hey @rvdeijk i believe changing the configuration in /etc/motioneye/motion.conf to turn tcp off is the answer to your problem,

Maybe a dumb question. But how do I get to that file: etc/motioneye/motion.conf
I thought that location should be on the sd card where motioneye is installed, but it's not. (I unplugged the sd card from raspberry pie and read it with a card reader) Can it be reached through the network? (windows File Explorer)?

hexordecimal, 2019-11-20:
@rvdeijk if you have mounted the sd cards partitions on linux via fdisk, then usually the 1st partition will be your MBR, 2nd will be your file system which is where you would find /etc/motioneye/motion.conf. With that said if youre are viewing the partitions on windows it is two different formats and not readable. If you have a windows machine, possibly try a live distrobution on your current pc with the card in the computer. Then once youre booted and connected to the network google references on mounting sd cards with linux(whatever live distrobution you decide to use). I'll post one of an image you can download if you're still having issues.
# This is the software you will need to turn an image into a bootable usb drive
https://www.linuxliveusb.com/
# This is a live distrobution of linux known as debian(choose the image according to x32 vs x64
https://www.debian.org/CD/live/
# or follow this example
https://www.howtogeek.com/howto/linux/create-a-bootable-ubuntu-usb-flash-drive-the-easy-way/
hope that helps

rvdeijk, 2019-11-28:
@hexordecimal Thx I was able to add the **netcam_use_tcp off** line into /etc/motioneye/motion.conf and now my stream is between 15 and 30fps instead of 4/5 fps 👍 

So thanks! No idea why this settings isn't turned off by default. Like more people should have this issue right.. It would be nicer to have an option in the UI to toggle between TCP on/off.
@ccrisan feature request?

alexmc1510, 2020-03-23:
Hello, If you consider this as new issue please close it and I will open a new one. My question is related Home Assistant motion eye addon. How can I modify this motion.conf file in MotionEye addon for home assistant? I am not able to find the folder.

Thanks

starbasessd, 2020-03-23:
/data/etc/
Most config files are located there.

As an aside, do a
cd /<enter>
find -name motion.conf
or whatever file you are looking for

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alexmc1510, 2020-03-23:
Thank you. Fogive me but I am completely new to this system. The only folders I can see through Samba addon in hassio are:
[image]
Where could I execute this command?

starbasessd, 2020-03-23:
At the console (keyboard and mouse attached) or by ssh into the Pi
What are you trying to accomplish?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

alexmc1510, 2020-03-24:
Thank you. Finally I was able to run this command but the file does not exist. I am running MotionEye addon into Home Assistant. The problem I am suffering is slow fps with a Reolink rtsp UDP camera.

Thanks

starbasessd, 2020-03-24:
Why are you trying to modify the file? What setting are you trying to get
to?
I can do my best to help, but need more specifics.
What device are you running HASSIO on?
How is the camera connected to the device? WiFi or Ethernet?
If you install motionEyeOS, or Raspbian and motionEye, and test, do you get
a faster/more acceptable framerate?
I get 3-5 frames per second (each) for 4 cameras, running motionEye on
Ubuntu, on a PC virtual machine.
I can get up to 10 frames per second currently on a Pi3B+ running
motionEyeOS for a single camera.
Pi3B+ and Ubuntu.motionEye are on Gigabit Ethernet, and the cameras are on
802.1`1n WiFi.
I will build a setup similar to yours and test as soon as you answer the
questions above.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jsxvenom, 2020-08-02:
Hi 
I am experience the same problem when connecting DAhua IP Camera IPC-HFW1320S-W to Raspberry Pi 3B+ with motioneyeos. I just testing motioneyeos for my plan to secure Avocado fruit tree from theft this week. I would try several IP camera that will suit my needs. 
I have a friend who should CCTV system so I borrow from him this DAhua IP camera, the price is expensive compare to my needs, but I need an IP camera to test the motioneyeOS. 
It runs and having very slow response using RTSP stream, after reading this issue, and otherink on internet I decide to go to DAhua Setting and edit the Video Stream setting.
I assume that the Raspberry PI and motioneyeos cannot handle so much data that have sent by the dahua IP camera
I  decrease the resolution from 2304*1296 to 1280*720 and the frame rate from 25 Fps to 5 Fps
when I got back to the motioneyeOS this setting work and I can see the video with 2 second delays.
After testing the dahua Ip camera I conclude that this IP camera have internal stand alone system that is useless to stream to motioneyeOS because this camera have all config that similar to motioneyeOS, and it is better to use it alone.
I think it is better for me to buy an USB camera or simple IP camera without many feature on the IP camera.
this is my experience..
thanks

jackkamm, 2021-08-01:
I also experienced laggy RTSP with a camera in motioneyeos, while RTSP stream in VLC was fine. In my case, the cause seemed to be that the camera FPS was much higher than the `Video Device->Frame Rate` set in motioneyeos. Increasing the video device frame rate to match the FPS coming from the camera fixed the problem for me. Note this setting is _different_ than the `Video Streaming->Streaming Frame Rate` setting. Hope this is helpful to someone!

closed 2021-08-01 by starbasessd