motioneye-project/motioneyeos #2183: Motioneye can't upload MediaFiles to GDrive

StefanGose91, 2019-11-05:
Hi everyone,

i am trying to automatically upload my video files with motioneye to my GDrive.
I generated the token and motioneye can create folders in my account. Testing with "Test Service" also works fine.

The media files are stored locally on a external HDD this is working too, but they never get uploaded.
In the log file I cant even a see a try to upload them.

I already reinstalled motioneye using pip (current version 0.41), this has not helped either

Here is the log file and I will provide 2 pictures of my configuration.

regards, hope u can help me :)

[image]
[image]

```
Nov 05 14:22:53 camsrv motion[631]: [1:ml1:Camera1] [DBG] [ALL] motion_loop: (Em) Init post capture 1
Nov 05 14:22:53 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82203.703125
Nov 05 14:22:53 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82294.234375
Nov 05 14:22:53 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82076.007812
Nov 05 14:22:53 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81879.906250
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 85901.617188
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 85392.257812
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [INF] [ALL] motion_loop: Emulating motion
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [DBG] [ALL] motion_loop: (Em) Init post capture 1
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" stop 1'
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" movie_end 1 /var/motionextern
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [NTC] [ALL] motion_loop: End of event 12
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 85860.328125
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 87425.992188
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 83925.593750
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 83610.234375
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 83361.109375
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 83410.796875
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [INF] [ALL] motion_loop: Emulating motion
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [DBG] [EVT] exec_command: Executing external command '/usr/local/lib/python2.7/dist-packages/motioneye/scripts/relayevent.sh "/etc/motioneye/motioneye.conf" start 1'
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [NTC] [EVT] event_new_video Source FPS 2
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [INF] [ENC] ffmpeg_open vbr/crf for codec: 1
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [NTC] [ENC] ffmpeg_open Selected Output FPS 2
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [NTC] [ENC] ffmpeg_avcodec_log: Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [NTC] [EVT] event_newfile: File of type 8 saved to: /var/motionextern/cam1/2019-11-05/14-22-54.avi
Nov 05 14:22:54 camsrv motion[631]: [1:ml1:Camera1] [NTC] [ALL] motion_detected: Motion detected - starting event 13
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 83067.320312
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82837.789062
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82625.109375
Nov 05 14:22:54 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82464.398438
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 86689.062500
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 82467.656250
Nov 05 14:22:55 camsrv motion[631]: [1:ml1:Camera1] [INF] [ALL] motion_loop: Emulating motion
Nov 05 14:22:55 camsrv motion[631]: [1:ml1:Camera1] [DBG] [ALL] motion_loop: (Em) Init post capture 1
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81648.593750
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81589.132812
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81513.421875
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81817.281250
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81577.054688
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81599.750000
Nov 05 14:22:55 camsrv motion[631]: [1:ml1:Camera1] [INF] [ALL] motion_loop: Emulating motion
Nov 05 14:22:55 camsrv motion[631]: [1:ml1:Camera1] [DBG] [ALL] motion_loop: (Em) Init post capture 1
Nov 05 14:22:55 camsrv motion[631]: [1:nc1:Camera1] [DBG] [NET] netcam_image_read_complete: Calculated frame time 81526.671875

```

AndyMAu, 2019-11-07:
Is your location name on your Google drive exactly as written in your motioneye camera setting?

StefanGose91, 2019-11-07:
Hi and thanks for the answer.

Yes it is exactly the same and motioneye itself created the folder, but it does not upload media files to it.

StefanGose91, 2019-11-07:
I tried it now with a local ftp server and it does not work either.

Is there any extra package I have to install to upload media files?