motioneye-project/motioneyeos #2186: Can't see my image in motioneyeOS camera

bend94, 2019-11-06:
Hi

I installed motioneye os on pi zero w.
I boot, i take my brower and i am able to reach MotionEye HTTP interface.
My cam is directly connected on the pi zero w
I can see my cam appearings in motioneye os but without any image : there is a camera with a circle

i got these messages in /var/log/motioneye.log

2019-11-06 20:10:48: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2019-11-06 20:10:48: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED

please advise
thanks

mdPlusPlus, 2019-11-08:
Which camera are you using?

bend94, 2019-11-11:
Hi

i use a pi cam, i suspected my cam to be out of order
i ve just ordered a brand new on Amazon
i will see

thanks

mdPlusPlus, 2019-11-11:
Are you talking about the officical Raspberry Pi Camera Module (v2.1)?

bend94, 2019-11-11:
Yes the official camera connected on the dedicated cam connector

mdPlusPlus, 2019-11-11:
Well, this seems unlikely to be honest. The Raspberry Pi Zero (W) has smaller connector than that camera module. And while there adapters for "zero cams" to the big connectors of regular Pis, I don't know about any adapter doing the opposite, so from big regular cam to small Zero connector.

Are you sure you're not confusing something here?

jasaw, 2019-11-12:
Which motioneyeos version are you running? Which web browsers did you use?

bend94, 2019-11-12:
Amazon has just informed me that i received this cam in my mailbox

Unistorm Raspberry Pi Zero Camera Fisheye Grand Angle 5 MP 1080p Vision
Nocturne Module de caméra pour Raspberry Pi Zero W
<https://www.amazon.fr/gp/product/B074DY544H/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1>

hexordecimal, 2019-11-16:
did you enable the camera with raspi-config?