motioneye-project/motioneyeos #2187: Setting file storage to attached external hard drive on motioneye hub

jk101290, 2019-11-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: raspberry pi 4 and raspberry pi zero with correct sd images installed on each

### Camera

I am using the following type of camera: MMAL

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: pi 4 is on ethernet, zero on wifi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Seagate 2TB HDD

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3820694/motioneye.log)

Hi, I'm having trouble with getting my file storage set to my external Sea Gate 2tb hard drive. It is attached to a raspberry pi 4 that is acting as a motioneye server hub (no camera attached.) I have a separate raspberry pi zero w with a camera attached that is added to the raspberry pi 4 hub.

On the server hub I've entered the command in ssh "df -h" and verified my hard drive is listed in /dev/sda1 mounted on /data/media/sda1

Looking in the file smb.conf it shows the following:
[global]
   workgroup = MOTIONEYE
   server string = motionEye
   security = user
   map to guest = bad user
   encrypt passwords = yes
   private dir = /var/lib/samba/private
   public = no
   writable = no
   include = /data/etc/smb.conf
   load printers = no
   printing = bsd
   printcap name = /dev/null
   disable spoolss = yes
   log level = 0
   syslog = 0
   preferred master = no
   domain master = no
   local master = no
   os level = 0

[sdcard]
   comment = SD Card Output Directory
   path = /data/output

[storage]
   comment = Attached Storage Devices
   path = /data/media

Not sure if I'm doing this right: in the file storage section on the pi zero, I'm trying to add the drive through network share. I've entered the IP of the server hub, the admin username and password (I tried "user" as well). For the Share Name field I've tried both MOTIONEYE and motionEye, and root path I've tried /data/media and /data/media/sda1 and /dev/sda1

The motioneye.log file always shows 

2019-11-07 11:58:04: [motioneye]    ERROR: failed to mount smb share "//192.168.1.85/motionEye" at "/data/media/motioneye_192_168_1_85_motioneye_user"

Any help would be greatly appreciated, thanks!

jk101290, 2019-11-08:
Figured it out

closed 2019-11-08 by jk101290

patraff, 2020-05-24:
jk
I want to do this as well. How did you come to a solution?
Pat

aerosolalchemy, 2020-05-24:
> jk
> I want to do this as well. How did you come to a solution?
> Pat

I think my problem was that there was some kind of conflict with Motioneye and Raspbian creating multiple directories to my drive. Somehow I ended up with different folders Seagate, Seagate 1, Seagate 2, etc. This was a while ago so I don't quite remember exactly, but I'm pretty sure that's what the problem was.

jk101290, 2020-05-24:
> jk
> I want to do this as well. How did you come to a solution?
> Pat

I think my problem was that there was some kind of conflict with Motioneye and Raspbian creating multiple directories to my drive. Somehow I ended up with different folders Seagate, Seagate 1, Seagate 2, etc. This was a while ago so I don't quite remember exactly, but I'm pretty sure that's what the problem was.

seaniedan, 2021-03-18:
I found that I couldn't save to external USB formatted as FAT32. I kept seeing the following if I ran the mount command:

[root@front ~]# mount 
/dev/root on / type ext4 (ro,relatime,norecovery)
devtmpfs on /dev type devtmpfs (rw,relatime,size=378188k,nr_inodes=94547,mode=755)
proc on /proc type proc (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /dev/shm type tmpfs (rw,relatime,mode=777)
tmpfs on /tmp type tmpfs (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (ro,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/mmcblk0p3 on /data type ext4 (rw,noatime)
/dev/mmcblk0p3 on /home/ftp/sdcard type ext4 (rw,noatime)
/dev/mmcblk0p3 on /home/ftp/storage type ext4 (rw,noatime)
overlay on /usr type overlay (rw,relatime,lowerdir=/usr,upperdir=/data/usr,workdir=/data/.overlay-usr)
overlay on /var/log type overlay (rw,relatime,lowerdir=/var/log,upperdir=/data/log,workdir=/data/.overlay-log)
overlay on /var/lib type overlay (rw,relatime,lowerdir=/var/lib,upperdir=/data/varlib,workdir=/data/.overlay-varlib)
/dev/sda1 on /data/media/sda1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sda1 on /home/ftp/storage/sda1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdb1 on /data/media/sdb1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdb1 on /home/ftp/storage/sdb1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdc1 on /data/media/sdc1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdc1 on /home/ftp/storage/sdc1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdd1 on /data/media/sdd1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdd1 on /home/ftp/storage/sdd1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
...
...
...
/dev/sdeh1 on /home/ftp/storage/sdeh1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdei1 on /data/media/sdei1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdei1 on /home/ftp/storage/sdei1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdej1 on /data/media/sdej1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdej1 on /home/ftp/storage/sdej1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdek1 on /data/media/sdek1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdek1 on /home/ftp/storage/sdek1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdel1 on /data/media/sdel1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdel1 on /home/ftp/storage/sdel1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdem1 on /data/media/sdem1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdem1 on /home/ftp/storage/sdem1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sden1 on /data/media/sden1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sden1 on /home/ftp/storage/sden1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdeo1 on /data/media/sdeo1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdeo1 on /home/ftp/storage/sdeo1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdep1 on /data/media/sdep1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdep1 on /home/ftp/storage/sdep1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdeq1 on /data/media/sdeq1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdeq1 on /home/ftp/storage/sdeq1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sder1 on /data/media/sder1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sder1 on /home/ftp/storage/sder1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
[... cut ...]
/dev/sdft1 on /home/ftp/storage/sdft1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdfu1 on /data/media/sdfu1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sdfu1 on /home/ftp/storage/sdfu1 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)

...I fixed the issue by formatting the external storage as FAT32 with GUID instead of MBR partition table.