motioneye-project/motioneyeos #2188: Device request: Banana Pi M2 Zero

mdPlusPlus, 2019-11-08:
So, the wiki says I should send it to you. However, I dont own such a device yet.
Could I just paypal/crypto/whatever the amount you need to buy one?

Are you even interested in making MotionEyeOS compatabile with the M2 Zero?

Device info: http://www.banana-pi.org/m2z.html  
Raspberry Pi Zero (W) form factor, but with a quad core CPU, so the framerate (hopefully) won't be restricted by it.

mdPlusPlus, 2019-11-13:
I guess the offer stands for you too, @jasaw. Are there any more active contributers that could be interested in such a task?

ccrisan, 2019-11-14:
Unfortunately they released yet another board with poor kernel support. According to [this wiki page](https://linux-sunxi.org/Sinovoip_Banana_Pi_M2_Zero#Mainline_kernel), there's no mainline kernel support and according to [this github page](https://github.com/BPI-SINOVOIP/BPI-M2Z-bsp), the board support package comes with kernel 3.4.

Long story short, given the painfully old kernel, we can't support this device. Thank you for offering to donate one, though.

Let's keep this issue open and see how things evolve with this device.

mdPlusPlus, 2019-11-14:
Oh, I wasn't aware of that. That's really a shame.  
Do you happen to know another SBC in that form factor (ideally with WiFi and the same camera connector) which uses a non single core CPU?

czadikem, 2020-03-09:
If @mdPlusPlus I get one I will see if I can get it to work and try to make an image for you as I want to do this to.

herwig-, 2020-07-19:
according to documents, there is now (since Apr) a version based on Kernel 4.4
http://wiki.banana-pi.org/Banana_Pi_BPI-M2_ZERO#Image_Release
does that help?

sagitt, 2021-04-12:
i hope the support for this. pi zero w have very old hardware