motioneye-project/motioneyeos #2189: pi4 hdmi preview

Woody565, 2019-11-08:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Latest

### Board Model

I am using the following board/model: Pi4).

### Camera

I am using the following type of camera: Fast network Cam

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

Hi All,

I am trying to get the HDMI preview to work but cannot get a picture over HDMI.

Im running the Pi4 with latest MEOS. I have tried using a PiZero and it works straight away.

Any Ideas?

Many Thanks

Dan

ccrisan, 2019-12-01:
Works fine here with 20190911.

jasaw, 2019-12-01:
HDMI preview is not available when fast network cam is in RTSP mode.
If HDMI preview is required for RTSP as well, I'll look into adding it.

ccrisan, 2019-12-01:
@jasaw good to know.