motioneye-project/motioneyeos #2190: Need help with Raspberry pi camera MotionEye problem

lolleri200, 2019-11-08:
So my MotionEye setup works fine, but after like 24 hours everyday, when I go connect the IP address it won't connect for some reason... And I can see that the Raspberry Pi and the Camera is working, but I can't connect to it using MotionEye! At first I thought it might be a heat issue, but I have now installed heat pads in my Pi and the Camera, but the issue is not fixed..

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

mdPlusPlus, 2019-11-14:
Which Pi model are you using? You didn't fill out the issue template.

lolleri200, 2019-11-15:
@mdPlusPlus Sorry now I updated all.

jasaw, 2019-11-17:
@lolleri200 I presume you have DHCP configured on your network. How long is your DHCP lease? 24 hours? Make sure your router hasn't assigned a different IP address to it after 24 hours. Log in to your router and check the Pi's IP address.