motioneye-project/motioneyeos #2191: Google Drive - Sign in with Google temporarily disabled for this app

skeggman, 2019-11-10:
I'm getting the following error message when I try to obtain the Google Drive Key:

[image]

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911.

### Board Model

I am using the following board/model: Raspberry PI 2B+.

### Camera

I am using the following type of camera: 1xMMAL, 1x Network Camera.

My camera model is: .

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

n/a

### Log Files

I consider the following log files relevant to this issue:

n/a

SteveSJLee, 2019-11-10:
Experiencing same issue regarding google drive. 
Turning on Less secure app access did not resolve this issue.

ccrisan, 2019-11-11:
Guys, a few days ago, I received a mail from Google asking me for some kind of verification of the motionEye (Google Drive) app. They were asking for a homepage of the project and I gave them the github link. One day later I got an answer from them:

[image]

Long story short, they insist for a domain to verify. And no, I can't mark the application as *Internal* because that option is missing, motionEye not being part of an organization.

I'm out of ideas and open to suggestions, as long as they don't involve acquiring and maintaining a domain for motionEye.

bedny22, 2019-11-12:
I hope you will find solution, just downloaded motion eye today and when i was setting it up i found out the google drive problem :( otherwise its really nice and useful.

digiltd, 2019-11-14:
Would be happy to get you a domain, knock up a page and host it for you (https)

SteveSJLee, 2019-11-14:
I wonder if they accept github pages as a vaild domain for them. It would be easier to create and maintain in my opinion. https://pages.github.com/

ccrisan, 2019-11-14:
@digiltd thanks for your offer! Let me first try to apply again with a small dummy page hosted on github pages as suggested by @SteveSJLee and see if they accept it.

digiltd, 2019-11-14:
Agreed, I did initially think the "not be a link that leads to: Third party hosting platform..." meant that a github page would not be an option, but that was down to my lazy reading :)

hexordecimal, 2019-11-16:
So did they take the motioneye app down too on the google play store or apple?

ccrisan, 2019-11-19:
Just a small update: I'm still waiting for the verification on https://ccrisan.github.io/motioneyeos/, which is a GitHub Pages-based website for motionEye, as recommended by @SteveSJLee.

The developers console page says "Your consent screen is being verified. This may take up to several weeks. Your last approved consent screen is still in use."

SteveSJLee, 2019-11-20:
@ccrisan Thanks for the update. Hope google approves it.

Jaizan, 2019-11-20:
I now have the same problem with google.     I have been using it OK for weeks, but today I tried to set up a different camera on a Pi Zero W and had this error.   I still have a different Pi saving data to a different folder on the same Google account, but cannot set up the new one.    Going to try Dropbox.

Jaizan, 2019-11-20:
I have the same error.    One Raspberry Pi has been running for weeks and continues to work OK.   I've just tried to put a different camera on a second Raspberry Pi and encountered this error.
Anyone with a currently working system is best advised not to delete and replace camera settings at present !

ccrisan, 2019-11-20:
The validation for https://ccrisan.github.io/motioneyeos/ failed: i need to prove ownership of the domain, which I obviously can't. It looks like we'll have to set up a domain.

I just bought `motioneye.org` and will set it up as a CNAME or something to `ccrisan.github.io`, then resubmit for verification. Let's hope third time's a charm.

ccrisan, 2019-11-21:
Guys this is slowly becoming a nightmare. After buying, registering and setting up http://www.motioneye.org/, ensuring that we have a *privacy policy* document linked from the homepage, Google wants me to provide a youtube video, demonstrating all this sh*t:

[image]

I'll try to put together such a video, in the hope that this is the last step in bringing back Google Drive services to motionEye.

PeterPedro, 2019-11-22:
wow

davidashdown, 2019-11-23:
Thank you in advance! :)

digiltd, 2019-11-25:
But have a look at what this guy does for the Google Drive Backup service for HomeAssistant. 

Not sure if it is any use, I know it is a different language etc but you might get something from the process he describes to both the users and the details he includes to keep Google happy. 

[image]

https://github.com/sabeechen/hassio-google-drive-backup/blob/master/hassio-google-drive-backup/AUTHENTICATION.md

https://philosophyofpen.com/hassiodrivebackup/

ccrisan, 2019-11-26:
The verification has failed again. This time, the problems were:

1. The YouTube video indicated that "localhost" was used instead of a production URL, which is apparently unacceptable, being considered a testing/staging domain. I will make a video using my own website, in hope that it satisfies the non-testing/staging condition.

2. The video showed an unsuccessful demonstration of the functionality, due to the blocked Google Drive situation. They have offered though to temporarily whitelist an (my) account for demonstration purposes.

I'm kind of looking forward to the next challenges already.

donvonhill, 2019-11-26:
Hi ccrisan, many thanks for your commitment and effort to fix functionality. Motieneyeos is an excellent tool and will definitely be able to wade through the pitfalls of googlu :-)

HairyCream, 2019-11-26:
Calin,
A big thanks for persisting with this ... it's a great project. : )

cardyology, 2019-11-27:
Thank you for trying to resolve this. I reinstalled an old build for the sheer hell of it last week and encountered this issue. 

Following.

ccrisan, 2019-11-28:
They rejected us again saying that the video showed an "unverified" domain (the video was shot from my own motionEye cams domain). They also stated that the video did not showcase how motionEye uses the [Google Drive] scope in regards to its definition.

I used motioneye.org (supposedly considered verified by now) and shot a longer video clearly demonstrating how Google Drive upload is enabled and configured and how videos are uploaded to Google Drive. Let's hope this is the last one.

sanchits2003, 2019-11-28:
Thank you for persisting with this verification nightmare, it is truly appreciated!

josecobi, 2019-11-28:
> They rejected us again saying that the video showed an "unverified" domain (the video was shot from my own motionEye cams domain). They also stated that the video did not showcase how motionEye uses [...]

Thank you ccrisan for all your effort. I wish I could help, but I am not tech savvy.

ijosh-development, 2019-11-29:
Thank you so much for your patience and effort!

Firerouge, 2019-11-30:
Consider putting an option under expert settings to override the google [client ID and secret](https://github.com/ccrisan/motioneye/blob/b7007ae83cb424bdf2257db85f8f278267cd7ead/motioneye/uploadservices.py#L150-L151) with the user's own.

TheHaf, 2019-11-30:
I think that it is a good option to override the client ID and secret. As sign in is still disabled, I set up rclone to sync with Google Drive. They've got the process of obtaining a client ID and secret written down and it was easy enough for me to follow.

reparator, 2019-12-01:
Hi Calin,
thanks a lot for your efforts in this issue and keep a stiff upper lip!
Now we see, who rules the world...

looxonline, 2019-12-02:
> Consider putting an option under expert settings to override the google [client ID and secret](https://github.com/ccrisan/motioneye/blob/b7007ae83cb424bdf2257db85f8f278267cd7ead/motioneye/uploadservices.py#L150-L151) with the user's own.
> 
> To suggest a temporary workaround for users, from the google developer console, create an OAth 2.0 client id and secret, then use the api library to switch on the photos/drive services.
> 
> With these you can recompile with your credentials [in the source](https://github.com/ccrisan/motioneye/blob/b7007ae83cb424bdf2257db85f8f278267cd7ead/motioneye/uploadservices.py#L150-L151)
> 
> or
> 
> connecting to the ssh of a device which has debugging enabled, and running the following two commands, will replace the current credentials with yours after you also reboot
> 
> `sed -i -e 's/jjqbWmICpA0GvbhsJB3okX7s/`_yourClientSecretHere_`/g' /usr/lib/python2.7/site-packages/motioneye/uploadservices.pyc`
> 
> `sed -i -e 's/349038943026-m16svdadjrqc0c449u4qv71v1m1niu5o.apps.googleusercontent.com/`_yourClientIDHere_`/g' /usr/lib/python2.7/site-packages/motioneye/uploadservices.pyc`

Tried this but to create the client ID and secret with the drive API requires approval by google which is exactly the same issue that is the root of this thread.

ccrisan, 2019-12-02:
Great news, everyone!

    
[image]

Thank you for your patience and your support. Please verify yourselves that this works indeed and let's close this damn issue.

donvonhill, 2019-12-02:
Many many thx!! Looks it works now! I do not want it to sound ungrateful and I do not want to upset you after all the fight, so I ask carefully :-) .. Will you apply this procedure also for google photo?

cinnamonsweets, 2019-12-02:
Is it just me or is uploading to google drive still not working? At the very least it creates a folder in google drive when I click Test Service. I checked local storage folder to find that the camera is still recording videos and taking pictures.

shane-powell, 2019-12-02:
Thanks it's working for me now. Didn't save the upload settings the first two tries but worked the third time. Both my cameras are now set up and uploading.

sabavnk, 2019-12-03:
Thank you very much @ccrisan, it works now. Really appreciate your support.

jakepals, 2019-12-03:
@ccrisan thank you for everything you have put yourself to ensure we are provided the best of service. A software very popular, especially among Wyze users, we are very, very grateful.

ccrisan, 2019-12-03:
Unfortunately, I think we'll have to go through the same process again, for Google Photos, as @donvonhill pointed out. What's even worse is that the validation process is probably going to take down the Google Drive service as well, as far as I can tell from Google's developers console, since it's basically the same project.

PeterPedro, 2019-12-04:
Yes, it's working now, hopefully the process will be faster now that you have all the requirements.

ccrisan, 2019-12-16:
The Google Photos scope has been approved as well. Everything should be working fine now.

closed 2019-12-16 by ccrisan

donvonhill, 2019-12-16:
Fantastic job ccrisan!! Many thanks for your effort!!

josecobi, 2019-12-16:
Awesome. Thank you very much!

chunkysteveo, 2020-01-09:
Just an FYI to anyone - I wasn't aware about this issue being raised and mine was still not uploading. Testing the Google Drive link it would time out. Needed a reboot of my camera and to obtain a new key which seemed to fix things. Thanks for your efforts @ccrisan !