motioneye-project/motioneyeos #2192: notification email failed error no 97 address family not supported by Protocol

cbrtumkur, 2019-11-12:
[image]
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi3-20190911.img.xz

### Board Model

I am using the following board/model: (Raspberry PI 3B+).

### Camera

I am using the following type of camera: (raspberry pi 2).

My camera model is: (raspberry pi 2).

### Network Connection

My motionEyeOS unit is connected to the network via: (Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

Dewey3, 2019-11-13:
You might want to try dropping the "@gmail.com" from the SMTP Account name and giving it another try.

cbrtumkur, 2019-11-14:
Sir I  Hotmail, the same error, thank you in advance

closed 2019-11-14 by cbrtumkur