motioneye-project/motioneyeos #2194: Camera in Auto-brightness loop

kittentakara, 2019-11-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: Latest version

### Board Model

Raspberry PI 3B+

### Camera

V4L2

### Network Connection

Wifi

### Peripherals

NASBOX network storage

Hey,

I have been trying to get MotionEyeOS to work properly for couple of weeks now. It one of the two USB camera's I have connected to my Rasp3B+ keeps changing brightness nonstop. I have turned off all the settings on MotionEye but it still seems to be in this endless brightness adjustment loop. I am guessing this is to do with Internal camera auto-brightness, but is there a way to turn this off?  
thanks

hexordecimal, 2019-11-16:
havent had this problem with my raspi cameras, i think v4l2 is the driver for the camera youre using on raspberry pi not the camera itself, i have 3 different raspberry pi cameras all different.

kittentakara, 2019-11-19:
Do think there is way to stop this? like a way to turn off the auto brightness setting on camera using drivers or such?

hexordecimal, 2019-11-20:
look up something about exposure settings for v4l2 [for example](https://gist.github.com/justinledwards/cc0c40ce611efc00346f)