motioneye-project/motioneyeos #2195: Notification email failed: timed out!

cbrtumkur, 2019-11-14:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

Vilquid, 2022-02-05:
### Preliminary Docs
I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: 20200606.

### Board Model
I am using the following board/model: Raspberry PI 3B+.

### Camera
I am using the following type of camera: V4L2

My camera model is: [Aukey Webcam Full HD 1080p](https://www.ldlc.com/fiche/PB00394614.html)

### Network Connection
My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals
I am not using others peripherals.

### Log Files
I consider the following log files relevant to this issue:

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/8008204/motioneye.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/8008200/boot.log)

starbasessd, 2022-02-05:
ERROR: failed to send mail: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/?p=BadCredentials v18sm1790178wrm.105 - gsmtp') indicated you don't have correct username or correct password for the account being used. The correct setup is:
[image]
I use the same email address: me@gmail.com for Email Address, SMTP Account, and From Address. For gmail, I use SMTP Port 587, and Use TLS ON. (Attached Pictures Time span is up to you, I never use more than 2-5 seconds...

Vilquid, 2022-02-05:
So, the correct completion is ? :
Email addresses : AAA@gmail.com
SMTP Server : smtp.gmail.com
SMTP Port : 587
SMTP Account : BBB@gmail.com
SMTP Password : password_of_BBB@gmail.com
From Address : CCC@gmail.com
Use TLS : ON
Attached Pictures Time Span : 3

starbasessd, 2022-02-05:
That should work, but the from & SMTP names are usually the same...

Vilquid, 2022-02-05:
I have this error message :
https://support.google.com/mail/?p=BadCredentials ay3sm12137252wmb.44 - gsmtp!
I don't know how it can help me because password_of_BBB@gmail.com and BBB@gmail.com are correct.

Email addresses : [AAA@gmail.com](mailto:AAA@gmail.com)
SMTP Server : smtp.gmail.com
SMTP Port : 587
SMTP Account : [BBB@gmail.com](mailto:BBB@gmail.com)
SMTP Password : [password_of_BBB@gmail.com](mailto:password_of_BBB@gmail.com)
From Address : [BBB@gmail.com](mailto:CCC@gmail.com)

starbasessd, 2022-02-05:
There are issues with some 'special' characters in the password in motionEye & motionEyeOS. And you don't tack on the "@gmail.com" to the password, I hope. 
If you  are using special characters, (depending on which specific ones) you may be able to edit the /data/etc/camera-X.conf file and place it directly in there.

Vilquid, 2022-02-05:
Yes I don’t tack @gmail.com. The problem is I don’t use special characters in my password. I gonna try to place my passeword in the file.

starbasessd, 2022-02-05:
Also, please try dev20201026.

Vilquid, 2022-02-05:
I tried to modify the /data/etc/camera-X.conf file but there is no file named camera-X.conf on the micro sd card. I also tried the dev20201026 version but Wifi doesn't work.

starbasessd, 2022-02-05:
The same wpa_supplicant.conf file that works on 20200606 works on dev20201026.
You need to edit it while it's up and running (SSH or Console) or use a Linux to mount and edit the file.