motioneye-project/motioneyeos #2197: Pimoroni Pan Tilt Hat where to save Action Button files?

alphanumeric007, 2019-11-15:
Yes, I confirm that I have read the CONTRIBUTING guide before opening this issue.

Yes, I confirm that I have read the FAQ before opening this issue.

motionEyeOS Version is 0.41

I am running motionEyeOS version: 20190911.

I am using the following board/model: Raspberry PI 3A+).

I am using the following type of camera: MMAL, 

My camera model is: Raspberry Pi V2

My motionEyeOS unit is connected to the network via: WiFi.

I am using the following peripherals that I consider relevant to this issue:
Pimoroni Pant Tilt Hat Full kit.

 
I am trying to add my Action Button files to control my Pan tilt Hat to my SD card but can't find the location I'm supposed to save them too? I put my card in a card reader and plugged that into my Pi 4B usb port.
The guide says /etc/motioneye/ but I cannot find an etc/motioneye folder? And there are several etc folders? Which drive / partition do I look at, and do I have to create the motioneye folder myself? Any help or guidance on this will be greatly appreciated. If its in the FAQ I missed it somehow, I did look there first.
https://github.com/ccrisan/motioneye/wiki/Action-Buttons

hexordecimal, 2019-11-16:
i dont believe it is located on the user interface, you would have to dive into your shell prompt and modify or create scripts accordingly.

hexordecimal, 2019-11-16:
can you even see the http user interface?

alphanumeric007, 2019-11-16:
I can see the WEB GUI no problem, and I can SSH into it. If I put an up_1 file in /data/etc I get the UP button in the WEB GUI. 
nano /data/etc/up_1 done from putty
Putting my python pan tilt code in that file just gets me a "An error has occurred, refreshing is advised" error massage though. If I put a link to an actual python file in it sort of works.
/usr/bin/python /data/etc/up.py
And up.py is as follows
import pantilthat
pantilthat.pan(0)
pantilthat.tilt(+7)

Now if I click the button the Pan Tilt moves, sometimes? I may have to click the button several times before something happens? Or there is a several second delay before it happens. I'm close but missed something?

alphanumeric007, 2019-11-16:
This is what I'm trying to follow
[  https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons  ](url)

alphanumeric007, 2019-11-16:
I redid it all over again and still got iffy reactions from the buttons. Even when it moved it didn't move all the way. On a hunch I edited my py files, the ones with the pan tilt python code in tem.
At the top I added "from time import sleep" and as a last line added "sleep(10)". 
I don't really understand why but it fixed it. Now its one click and it moves fully to where its supposed to go.

popoviciri, 2019-11-16:
Have a look here: https://github.com/ccrisan/motioneyeos/pull/2016

alphanumeric007, 2019-11-17:
I hunted and hunted for that info and struck out. Thank you for posting it, much appreciated. I clip and pasted that info into a text file to look at latter. For next time. I have it working now just the way I wanted. Took me a while with lots of trial and error but I finally got it. My Linux skills are NOOB level which doesn't help. 
Mine is setup to move to a specific preset viewing position when I clcik a button.  
Anyway, thanks again.

closed 2019-11-17 by alphanumeric007